letscontrolit/ESPEasy #1503: (Unintended?) multiple readouts from BME280

m-anish, 2018-06-13:
https://github.com/letscontrolit/ESPEasy/blob/f41f0969f0e5dc3d31767918ffaa8f1cdec2c23e/src/_P028_BME280.ino#L331

The above line of code causes multiple readouts of the BME280 sensor, every second or so, causing it to heat up and provide a false reading. A simple way to reproduce this is to have this sensor enabled and sending data to controller, and have sleep mode enabled. Then there are readouts every second or so.

TD-er, 2018-06-13:
That line is just plain wrong.
It should be like `timePassedSince(sensor.last_measurement) > BMx280_MEASUREMENT_INTERVAL_MSEC`

TD-er, 2018-06-13:
Could you test it? I'm not at a computer now that could compile stuff, nor write it to an ESP node.

m-anish, 2018-06-14:
Okay, this does two things:

1. It limits the the sensor read out rate to atleast once per 50 seconds - GOOD
2. The module has to stay on for atleast 50 seconds to sent it's first reading - NOT SO GOOD

It would be best if the first reading happened right away, and every subsequent reading waited atleast 50 seconds. 

So I'd say it's a partial fix.

TD-er, 2018-06-14:
OK, I will have a look at it.
It was also my intention to add some extra parameters for people to set the read interval and some (pre defined) matching filtering settings to make sure the filters match the read interval.

And I also just received the BME680, so maybe it could be included in this plugin also, to make it a BMxx80 ;) (it now does BMP280 and BME280)

Thanks for taking a closer look at this.

fluppie, 2018-06-27:
I detected the same, had awake from deep sleep on 30s and then 270s zzzzzz. No readings from BME280 but got some from the TSL lux sensor and the analog input. Increased to 60s, works now :). 

`

Firmware
--
Build | 20102  - Mega
Libraries | ESP82xx Core 2_4_1, NONOS SDK 2.2.1(cfd48f3), LWIP: 2.0.3
GIT version |  
Plugins | 47 [Normal]
Build Md5 | 4d44355f4d44355f4d44355f4d44355f
Md5 check | fail !
Build time | Jun 20 2018 18:01:35

`

stefbo27, 2018-07-01:
Having short awake times after deep-sleep <10s I can confirm that BME280 is not read. My workaround is to uncomment the check for timeOutReached. 
I guess after deep-sleep sensor.Last_measurement gets resret to 0 so that at least BMx280_MEASUREMENT_INTERVAL_MSEC have to pass until the first measurement is performed.
Therefore I checked the value of sensor.last_measurement with log functionality.
`  String log2;
  log2.reserve(120); // Prevent re-allocation
  log2 = sensor.getDeviceName();
  log2 += F(": last: ");
  log2 += sensor.last_measurement;
  log2 += F(" now: ");
  log2 += current_time;
  addLog(LOG_LEVEL_DEBUG, log2);
`
There I get values of either 1900 msec or 6100 msec each time
[image]

So either BME280 is called before and no log is written or sensor.last_measurement is not initialized properly. 
Stefan

TD-er, 2018-07-01:
There are more issues with the BMx280 plugin, which I'm working on right now.
For example between waking the sensor and taking a reading, the plugin is also waiting and thus blocking the entire node.

This also affects timing critical tasks like the one performed in 10_per_second and 50/sec.
So I am looking into this plugin. (and also at other plugins that may do similar things)

Things to fix/verify:

- Make shorter measurement intervals possible
- Only perform the minimum amount of measurements needed, based on the active filter (in the sensor)
- Make the sensor go to sleep again.
- Do not occupy the entire system when waiting for a stable sample.

TD-er, 2018-08-29:
@fluppie Does that also happen on the latest builds?
I do use some filter and have thus a minimum amount of samples to wait for the effects of this filter to minimize.
But I can ofcourse change this filter value, or increase the number of read samples.

fluppie, 2018-08-29:
Ok weird, latest GIT build (using Atom and PlatformIO) doesn't subscribe to MQTT and also the emonCMS plugin doesn't work.

> 
INIT : Booting version:  (ESP82xx Core 2_4_2, NONOS SDK 2.2.1(cfd48f3), LWIP: 2.0.3)
102 : INIT : Warm boot #1 - Restart Reason: External System
105 : FS   : Mounting...
130 : FS   : Mount successful, used 75802 bytes of 957314
139 : CRC  : No program memory checksum found. Check output of crc2.py
171 : CRC  : SecuritySettings CRC   ...OK 
275 : INIT : Free RAM:26888
276 : INIT : I2C
277 : INIT : SPI not enabled
293 : INFO : Plugins: 47 [Normal] (ESP82xx Core 2_4_2, NONOS SDK 2.2.1(cfd48f3), LWIP: 2.0.3)
294 : WIFI : Set WiFi to STA
326 : WIFI : Connecting WiFi2016AP1 attempt #0
1107 : BMx280 : Detected BME280
1661 : WD   : Uptime 0 ConnectFailures 0 FreeMem 25504
4074 : BME280: dew point 15.61C
4087 : BME280 : Address: 0x77
4087 : BME280 : Temperature: 22.56
4087 : BME280 : Humidity: 64.94
4088 : BME280 : Barometric Pressure: 1018.31
4229 : WIFI : Connected! AP: WiFi2016AP1 (78:8A:20:5A:41:76) Ch: 1 Duration: 3790 ms
4230 : WIFI : DHCP IP: 192.168.1.54 (ESP-Easy-123) GW: 192.168.1.1 SN: 255.255.255.0   duration: 23 ms

> 

I waited for more than 5 minutes, reading interval was set to 30s. Also tried 15s and 60s. Will rebuild with older SDK and report tomorrow.

TD-er, 2018-08-29:
Problems with EmonCMS were also reported here:
https://github.com/letscontrolit/ESPEasy/issues/1684#issuecomment-417107197

And also check the serial log level. When set to "none" it will only output until wifi connection has been made.  (I still should remove that part...)

TD-er, 2019-10-10:
Is this still an issue?

closed 2019-10-27 by TD-er