letscontrolit/ESPEasy #1508: Implementation of AS3935 lighting sensor

talion303, 2018-06-17:
Feature request for AS3935 lighting sensor

### Expected behavior
AS3935 could be connected via I2C or SPI. Configuration of the registers of the AS3935 can be done via webinterface. 

### Actual behavior
AS3935 is not supported.

There are libs out there, for Example:

https://github.com/raivisr/AS3935-Arduino-Library

misko903, 2018-11-18:
Would be great add-on to my weather station

talion303, 2019-01-20:
Found unfinished Plugin:

https://github.com/bertbigb/_P214_AS3935-LightningSensor-MOD1016/blob/master/AS3935-LightningSensor-MOD1016.ino

laserrichi, 2021-08-10:
Hi, what is the status quo, when will it be implemented? I would like to use the sensor.

TD-er, 2021-08-10:
I just checked my orders at AliExpress and just realized that the part is already a year in my drawer.
So for now that's the only progress so far... ;(

PeJeWe, 2023-11-08:
what is the status , has is it bein implemented? I would also like to use the sensor.

tonhuisman, 2024-12-27:
This sensor is implemented in [P169](https://espeasy.readthedocs.io/en/latest/Plugin/P169.html#p169-page) since ~ June 2024
Issue can be closed.

closed 2024-12-27 by TD-er