letscontrolit/ESPEasy #1510: Sensor is sending every second during the entire awake period icm deep sleep

dannybloe, 2018-06-18:
Trying version ESP_Easy_mega-20180615_normal_ESP8266_4096 with one sensor (analog pin) with a delay of, let's say, 30s reporting to Domoticz.

When the ESP is set to deep sleep with an awake period of xx secondes, when it awakes, it is keeps sending the sensor data to Domoticz every second during the awake period. 

I was expecting that the sensor would send only once during the awake period or respect the delay setting of the sensor (then I would set the delay longer than the awake period). So, just to be sure that the ESP activates properly I have set an awake time of 6 seconds but then I get around 6 updates in Domoticz (causing all kind of event scripts to be activated 6 times as well).

Cheers,
Danny

TD-er, 2018-06-18:
There have been more reports about his.
It is a known (and not yet fixed) issue, at this moment.

fluppie, 2018-06-27:
I thought this was "on purpose" to push updated values in case the connection wasn't ok or with really low RSSI values.

dannybloe, 2018-06-27:
well.. I hope not :) You could always check the response value of the http-request but I surely don't want a zillion request being made. Then perhaps make a setting or so.

fluppie, 2018-06-27:
This is what happens during a 90s "awake time" from deepsleep:

`
>> NetworkError when attempting to fetch resource. <<
9637: EVENT: TSL2561_14#Broadband=27151.00
10428: ADC : Analog value: 1015
10429: EVENT: ANA_14#Analog=1015.00
11654: WIFI : DHCP IP: 192.168.1.49 (Keuken-test-14) GW: 192.168.1.1 SN: 255.255.255.0 duration: 5983 ms
11665: EVENT: WiFi#Connected
11670: Webserver: start
11672: MQTT : Intentional reconnect
11713: MQTT : Connected to broker with client ID: ESPClient_5C:CF:7F:10:4D:C9
11715: Subscribed to: /Keuken_test/#
11717: EVENT: MQTT#Connected
11727: WD : Uptime 1 ConnectFailures 0 FreeMem 17944
11732: SYS : -84.00
11733: EVENT: RSSI_14#=-84.00
13733: Timeout while reading input data!
14170: TSL2561: Address: 0x39: Mode: 2: Gain: 0: Lux: 3072.16: Infrared: 14589.00: Broadband: 28762.00
14172: EVENT: TSL2561_14#Lux=3072.16
14175: EVENT: TSL2561_14#Infrared=14589.00
14176: EVENT: TSL2561_14#Broadband=28762.00
15854: Timeout while reading input data!
15890: ADC : Analog value: 1009
15891: EVENT: ANA_14#Analog=1009.00
18091: SYS : -84.00
18093: EVENT: RSSI_14#=-84.00
20105: Timeout while reading input data!
20540: TSL2561: Address: 0x39: Mode: 2: Gain: 0: Lux: 2796.09: Infrared: 13949.00: Broadband: 27106.00
20542: EVENT: TSL2561_14#Lux=2796.09
20545: EVENT: TSL2561_14#Infrared=13949.00
20546: EVENT: TSL2561_14#Broadband=27106.00
22227: Timeout while reading input data!
22262: ADC : Analog value: 1016
22264: EVENT: ANA_14#Analog=1016.00
24365: Timeout while reading input data!
24433: SYS : -84.00
24434: EVENT: RSSI_14#=-84.00
26483: Timeout while reading input data!
26921: TSL2561: Address: 0x39: Mode: 2: Gain: 0: Lux: 2793.49: Infrared: 13947.00: Broadband: 27096.00
26923: EVENT: TSL2561_14#Lux=2793.49
26926: EVENT: TSL2561_14#Infrared=13947.00
26928: EVENT: TSL2561_14#Broadband=27096.00
28613: Timeout while reading input data!
28648: ADC : Analog value: 1011
28650: EVENT: ANA_14#Analog=1011.00
30743: Timeout while reading input data!
30806: SYS : -84.00
30808: EVENT: RSSI_14#=-84.00
32862: Timeout while reading input data!
33300: TSL2561: Address: 0x39: Mode: 2: Gain: 0: Lux: 2795.15: Infrared: 13948.00: Broadband: 27102.00
33302: EVENT: TSL2561_14#Lux=2795.15
33305: EVENT: TSL2561_14#Infrared=13948.00
33307: EVENT: TSL2561_14#Broadband=27102.00
34991: Timeout while reading input data!
35022: ADC : Analog value: 1013
35024: EVENT: ANA_14#Analog=1013.00
37113: Timeout while reading input data!
37177: SYS : -84.00
37179: EVENT: RSSI_14#=-84.00
39233: Timeout while reading input data!
39671: TSL2561: Address: 0x39: Mode: 2: Gain: 0: Lux: 2802.02: Infrared: 13958.00: Broadband: 27135.00
39673: EVENT: TSL2561_14#Lux=2802.02
39676: EVENT: TSL2561_14#Infrared=13958.00
39678: EVENT: TSL2561_14#Broadband=27135.00
41355: Timeout while reading input data!
41389: ADC : Analog value: 1009
41391: EVENT: ANA_14#Analog=1009.00
43479: Timeout while reading input data!
43550: WD : Uptime 1 ConnectFailures 0 FreeMem 14112
[... 87 lines cut ...]
89034: EVENT: ANA_14#Analog=1013.00
91122: Timeout while reading input data!
91187: SYS : -85.00
91189: EVENT: RSSI_14#=-85.00
93250: Timeout while reading input data!
93685: TSL2561: Address: 0x39: Mode: 2: Gain: 0: Lux: 2866.94: Infrared: 14179.00: Broadband: 27622.00
93687: EVENT: TSL2561_14#Lux=2866.94
93690: EVENT: TSL2561_14#Infrared=14179.00
93692: EVENT: TSL2561_14#Broadband=27622.00
95406: Timeout while reading input data!
95437: ADC : Analog value: 1020
95439: EVENT: ANA_14#Analog=1020.00
97531: Timeout while reading input data!
97595: SYS : -85.00
97597: EVENT: RSSI_14#=-85.00
99658: Timeout while reading input data!
100093: TSL2561: Address: 0x39: Mode: 2: Gain: 0: Lux: 2889.27: Infrared: 14298.00: Broadband: 27849.00
100095: EVENT: TSL2561_14#Lux=2889.27
100098: EVENT: TSL2561_14#Infrared=14298.00
100100: EVENT: TSL2561_14#Broadband=27849.00
101780: Timeout while reading input data!
101814: ADC : Analog value: 1008
101816: EVENT: ANA_14#Analog=1008.00
>> NetworkError when attempting to fetch resource. <<
`

tanker11, 2018-06-27:
Hello,

I have exactly the same behavior on an earlier version, namely on mega-20180314_normal_ESP8266_1024. I am sending values on UDP, and those sent by every seconds during the awake period.

And I also thought it was on purpose so tried to find a related setting...
I am happy that it is already addressed.

RT

tonhuisman, 2022-05-07:
This seems resolved, so can probably be closed

closed 2022-05-07 by TD-er