letscontrolit/ESPEasy #1511: OTA feature causes bootloop

s0170071, 2018-06-18:
If I just turn on the OTA feature, (#define FEATURE_ARDUINO_OTA) I get a bootloop. I turn it off- loop is gone. 

```

INIT : Booting version: (custom) (ESP82xx Core ea4720b0, NONOS SDK 2.2.1(cfd48f3), LWIP: 2.0.3)
90 : INIT : Warm boot #67
92 : FS   : Mounting...
117 : FS   : Mount successful, used 75802 bytes of 957314
142 : CRC  : No program memory checksum found. Check output of crc2.py
rules1.txt 1
rules2.txt 0
rules3.txt 0
rules4.txt 0

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
vea4720b0
~ld
⸮U90 : 

INIT : Booting version: (custom) (ESP82xx Core ea4720b0, NONOS SDK 2.2.1(cfd48f3), LWIP: 2.0.3)
91 : INIT : Warm boot #68
92 : FS   : Mounting...
117 : FS   : Mount successful, used 75802 bytes of 957314
143 : CRC  : No program memory checksum found. Check output of crc2.py
rules1.txt 1
rules2.txt 0
rules3.txt 0
rules4.txt 0

```

TD-er, 2018-06-18:
I never tested this `FEATURE_ARDUINO_OTA` myself and based on the changelog of 2.0.0-dev11, it was giving problems before.

It looks like it starts some service on a specific port, but I guess with event-based wifi, such a service should only be started after wifi connection has been made.

s0170071, 2018-06-18:
Good point.

closed 2023-09-04 by TD-er