letscontrolit/ESPEasy #1512: Nodelist always empty

papperone, 2018-06-18:
### Summarize of the problem/feature request
Not sure if I missed some settings, but I've this anomaly an all my nodes.
I tried several nighly build from March until last one of yesterday and none of my devices shows anymore the "node list" in the Main Page. An old test device with DEV12 installed works perfectly.

### Expected behavior
List of all nodes in my network as usual...

### Actual behavior
Node List is empty

### System configuration
Tried on several Wemos D1 Mini and other selft build PCB all using ESP-12F chip

ESP Easy settings/screenshots: 
[image]
[image]

### Rules or log data
not applicable

Budman1758, 2018-06-18:
I assume you have a UDP port number defined in the advanced settings page? I have the latest mega running on 2 units and the node list is working as it should. I update those 2 units with every mega release.

papperone, 2018-06-18:
of course I have UPD port defined, the same port I've been using last 2 years...
only on older firmware I can see the nodes list, all the ones with ESPEasy mega does not show it!
is there any "not allowed" port number (I always used 65500)

Budman1758, 2018-06-18:
Don't know if any unallowed ports. I have been using the same port number all along too.  There have been times when the node list was not working or took a long time to work but as of right now it works fine on the 2 units I updated last night.

TD-er, 2018-06-18:
There could have been some signed/unsigned issue I am not aware of.
I use port 8003 on my nodes (not sure where that one came from) and it is working with all my nodes from v2.0 mega from around Jan 1st 2018 to the latest current build on an ESP32

papperone, 2018-06-18:
ok I think I got it, while deploying new firmware I change unit number adding 50 (1=51, 2=52, etc.) and it seems Node List only lists nodes up to 31; is there a specific limit for this that I overlooked?

Grovkillen, 2018-06-18:
Yes, only node numbers up to 31 is allowed for current UDP inter communication implementation.

Grovkillen, 2018-06-18:
https://www.letscontrolit.com/wiki/index.php/ESP_Easy_web_interface#Config_page

papperone, 2018-06-18:
ok, I will renumber all my nodes...
the major issue is that I'm planning to have more than 31 nodes in my system, is there a specific reason for this hard coded limit?

TD-er, 2018-06-18:
At this moment, there is a pre-defined array to hold these values of other ESP nodes.
I can assure you, that list will be dealt with, along with another big memory hog.

papperone, 2018-06-18:
ok thanks :)

Grovkillen, 2019-10-10:
I close this.

closed 2019-10-10 by Grovkillen