letscontrolit/ESPEasy #1513: generic http plugin slows down system

s0170071, 2018-06-18:
I got a plugin that measures how often the "tenpersecond" part of a plugin is actually called each second. It usually shows 10 unless I activate a generic http controller. There are three plug ins that are serviced by that controller each 60 seconds. 
Still the "tenpersecond" value drops from 10 to 6

TD-er, 2018-06-18:
Hmm that's a lot of edits and reading between the lines, trying to follow your thoughts ;)

Could you summarize the problem(s)?

s0170071, 2018-06-18:
Bottom line: message delay should be non blocking.

TD-er, 2018-06-19:
Since it is also mentioned on the wiki, as a solution, I was also thinking about doing something like that.

tonhuisman, 2022-04-18:
This seems to have been fixed, so issue can be closed.

closed 2022-04-18 by TD-er