letscontrolit/ESPEasy #1518: Ultrasonic Plugin _P013_HCSR04.ino broken with last commit

lhurt, 2018-06-19:
I use a self compiled version.

### Commit dcf8e86db75a69406871e9ac2fa8b2e414f7c3f1 (Jun. 9th) broke plugin _P013_HCSR04.ino
Build with tag _mega-20180606_

_ works and shows valid values like `"ULTRASONIC : Distance: 167.00"`.
Build with tag _mega-20180611_ doesn't work and shows `"ULTRASONIC : TaskNr: 2 Distance: No reading!"`.

### Expected behavior
The values should be retrieved again.

### Actual behavior
No values are retrieved making the sensor useless.

### Steps to reproduce
1. Build firmware with tag _mega-20180611_ or later
2. See log output

### System configuration
Hardware: H801 (ESP 8266)

Log of working version (_mega-20180606_)
[image]

Log of defective version (_mega-20180611_)
[image]

lhurt, 2018-09-13:
Is it so hard to fix this? I use this pretty heavily and I am stuck with an old version.

TD-er, 2018-09-13:
I will revert it to an older version and do fix it later.
I wanted to test it myself, with a test setup, but I have been busy lately with other issues and the last 2 days trying to figure out why MQTTimporter crashes (hard).

lhurt, 2018-09-17:
> I will revert it to an older version and do fix it later.
> I wanted to test it myself, with a test setup, but I have been busy lately with other issues and the last 2 days trying to figure out why MQTTimporter crashes (hard).

Thanks for the quick reply.

JojoS62, 2018-09-27:
ok, I think I found the reason: the module is killing itself. There is a temporary oject created and inserted into a std::map. But the object is deleted on leaving the PLUGIN_INIT:
```
INIT : Booting version:  (ESP82xx Core 2_4_2, NONOS SDK 2.2.1(cfd48f3), LWIP: 2.0.3)
100 : INIT : Cold Boot - Restart Reason: External System
102 : FS   : Mounting...
126 : FS   : Mount successful, used 76304 bytes of 957314
151 : CRC  : No program memory checksum found. Check output of crc2.py
179 : CRC  : SecuritySettings CRC   ...OK 
272 : INIT : Free RAM:27376
273 : INIT : I2C
273 : INIT : SPI not enabled
277 : constructor P_013_sensordef
4278 : delete P_013_sensordef
4279 : ULTRASONIC : TaskNr: 1 TrigPin: 12 IRQ_Pin: 15 max dist cm: 150 max echo: 8607 nr_tasks: 1
```
so the constructor is called and after that the destructor. A copy of the struct is still in the map but pointing to an deleted NewPingESP8266 object. That caused also firmware reboots.

JojoS62, 2018-09-27:
here is the construction of a temporay object: https://github.com/letscontrolit/ESPEasy/blob/30f5c5cb7dca661bdd4d5a51f242b6fbba94f624/src/_P013_HCSR04.ino#L110
I have uncommented the delete call in https://github.com/letscontrolit/ESPEasy/blob/30f5c5cb7dca661bdd4d5a51f242b6fbba94f624/src/_P013_HCSR04.ino#L24, then it works.
Maybe it is better to call delete in the PLUGIN_EXIT before the erase. Or not using a temporary object for inserting in the map.

JojoS62, 2018-09-27:
this will fix it in PLUGIN_INIT:
```
        P_013_sensordef *newSensordef = new P_013_sensordef(Plugin_013_TRIG_Pin, Plugin_013_IRQ_Pin, max_cm_distance);
        P_013_sensordefs[event->TaskIndex] = *newSensordef;
```
in PLUGIN_EXIT, the erase is called and this will call the destructor, so memory will be cleaned correctly.