letscontrolit/ESPEasy #1519: build_flags for esp8266_4M mistyped?

dmoibm, 2018-06-20:
I think there is a mistype in platformio.ini.

If I try to build only for normal_ESP8266_4096 I get the error: Nothing to build. Please put your source code files to…

Actual in platformio.ini row 134:
[esp8266_4M]
board_build.flash_mode    = dio
build_flags                         = ${common.build_flags} -Wl,-Tesp8266.flash.4m1m.ld
```
[esp8266_4M]

board_build.flash_mode    = dio

build_flags                         = ${common.build_flags} -Wl,-Tesp8266.flash.4m1m.ld
```
sp8266_4M]
board_build.flash_mode    = dio
build_flags               = ${common.build_flags} -Wl,-Tesp8266.flash.4m1m.ld
`

`[esp8266_4M]
board_build.flash_mode    = dio
build_flags               = ${common.build_flags} -Wl,-Tesp8266.flash.4m1m.ld`
[esp8266_4M]
board_build.flash_mode    = dio
build_flags               = ${common.build_flags} -Wl,-Tesp8266.flash.4m.ld

### Summarize of the problem/feature request

YOUR TEXT GOES HERE

### Expected behavior

YOUR TEXT GOES HERE

### Actual behavior

YOUR TEXT GOES HERE

### Steps to reproduce

1. 
2. 
3. 

### System configuration

Hardware:

ESP Easy version: 

ESP Easy settings/screenshots: 

### Rules or log data

```

```

TD-er, 2018-06-20:
Hmm, that's strange.
I've changed this file very recently, but I am sure I've built the "normal 4096" one a few times since.

I will look into this this evening.

I sometimes get this error when I change the platformio.ini file outside the Atom editor (e.g. Git rebase)
Then it has to generate a new cpp file in the background and as long as that's not finished, a request to build may fail.

dmoibm, 2018-06-20:
Sorry: with the newest Platformio Espressif 8266 Platform Release 1.7.1 it works with ...4m1m!
With the older release 1.5.0 it works onli with ...4m.

Thanks

TD-er, 2018-06-20:
OK, good to know :)

closed 2018-06-20 by TD-er