letscontrolit/ESPEasy #1523: Web Log drops lines

s0170071, 2018-06-20:
I log to the weblog once per second, there are now and then there are lines missing. The following log has lines starting with "MT681", followed by an incrementing integer. 

Look for entries 11 and 12. They are missing. As well as 23. And so on.
Log frequency is about 2 lines per second.

```
5794: MT681: 1 e2s:1476 s2s:1960 s2e:484 e2e:1960
7775: MT681: 2 e2s:1496 s2s:1980 s2e:485 e2e:1981
8766: MT681: 3 e2s:520 s2s:1005 s2e:471 e2e:991
11051: reading bytes: 0 1 0 0 0 11 3 4 8 > 0 0 0 0 0 0 178 142 value: 45710.00
11799: MT681: 5 e2s:567 s2s:1045 s2e:484 e2e:1051
12839: MT681: 6 e2s:552 s2s:1036 s2e:488 e2e:1040
13171: connect: 192.168.1.41 OK, sending read request: 0 1 0 0 0 6 3 4 120 55 0 2 
13251: reading bytes: 0 1 0 0 0 7 3 4 4 > 0 0 0 47 value: 47.00
13879: MT681: 7 e2s:557 s2s:1045 s2e:483 e2e:1040
14919: MT681: 8 e2s:556 s2s:1039 s2e:484 e2e:1040
15464: connect: 192.168.1.41 OK, sending read request: 0 1 0 0 0 6 3 4 119 53 0 4 
15651: reading bytes: 0 1 0 0 0 11 3 4 8 > 0 0 0 0 0 0 178 142 value: 45710.00
16864: MT681: 9 e2s:1475 s2s:1960 s2e:469 e2e:1945
17859: connect: 192.168.1.41 OK, sending read request: 0 1 0 0 0 6 3 4 120 55 0 2 
18050: reading bytes: 0 1 0 0 0 7 3 4 4 > 
18150: 0 0 0 45 value: 45.00
18819: MT681: 10 e2s:1487 s2s:1956 s2e:468 e2e:1955
24715: MT681: 13 e2s:1495 s2s:1979 s2e:466 e2e:1961
25284: connect: 192.168.1.41 OK, sending read request: 0 1 0 0 0 6 3 4 119 53 0 4 
25474: reading bytes: 0 1 0 0 0 11 3 4 8 > 0 0 0 0 0 0 178 142 value: 45710.00
26674: MT681: 14 e2s:1494 s2s:1960 s2e:464 e2e:1959
27686: connect: 192.168.1.41 OK, sending read request: 0 1 0 0 0 6 3 4 120 55 0 2 
27873: reading bytes: 0 1 0 0 0 7 3 4 4 > 0 0 0 47 value: 47.00
28634: MT681: 15 e2s:1495 s2s:1960 s2e:464 e2e:1960
30674: MT681: SellCounter: 6435.37  BuyCounter: 4392.68  Power: 363.00  callsPerSecond: 60.00 s2e:564 e2e:2040
30683: WD   : Uptime 1 ConnectFailures 0 FreeMem 17192
36470: MT681: 19 e2s:1496 s2s:1960 s2e:464 e2e:1960
38430: MT681: 20 e2s:1496 s2s:1960 s2e:464 e2e:1960
39489: connect: 192.168.1.41 OK, sending read request: 0 1 0 0 0 6 3 4 119 53 0 4 
39577: reading bytes: 0 1 0 0 0 11 3 4 8 > 0 0 0 0 0 0 178 142 value: 45710.00
41774: connect: 192.168.1.41 OK, sending read request: 0 1 0 0 0 6 3 4 120 55 0 2 
42065: reading bytes: 0 1 0 0 0 7 3 4 4 > 0 0 0 49 value: 49.00
42307: MT681: 22 e2s:1329 s2s:1897 s2e:484 e2e:1813
46207: MT681: 24 e2s:1496 s2s:1960 s2e:484 e2e:1980
46580: connect: 192.168.1.41 OK, sending read request: 0 1 0 0 0 6 3 4 120 55 0 2 
46766: reading bytes: 0 1 0 0 0 7 3 4 4 > 0 0 0 50 value: 50.00
48167: MT681: 25 e2s:1476 s2s:1960 s2e:484 e2e:1960
48876: connect: 192.168.1.41 OK, sending read request: 0 1 0 0 0 6 3 4 119 53 0 4 
49064: reading bytes: 0 1 0 0 0 11 3 4 8 > 0 0 0 0 0 0 178 142 value: 45710.00
51169: connect: 192.168.1.41 OK, sending read request: 0 1 0 0 0 6 3 4 120 55 0 2 
51264: reading bytes: 0 1 0 0 0 7 3 4 4 > 
51364: 0 0 0 50 value: 50.00
52060: MT681: 27 e2s:1467 s2s:1933 s2e:485 e2e:1952
53470: connect: 192.168.1.41 OK, sending read request: 0 1 0 0 0 6 3 4 119 53 0 4 
53565: reading bytes: 0 1 0 0 0 11 3 4 8 > 0 0 0 0 0 0 178 142 value: 45710.00
54039: MT681: 28 e2s:1495 s2s:1980 s2e:484 e2e:1979
57059: MT681: 30 e2s:555 s2s:1039 s2e:465 e2e:1020
```

Grovkillen, 2018-06-20:
Is the same thing happening in the serial log?

TD-er, 2018-06-20:
I assume you're running just 1 viewer?
If you use the inspect window in the browser and then the network tab, you can see the JSON requests.
There is also some TTL value in the JSON, which indicates the read interval to get the next one.
The current implementation of the log buffer uses a fixed number of lines. This means a line is lost if that line count is full before the JavaScript reads them.

giig1967g, 2018-06-20:
The old log viewer was less fancy but was much more reliable.
In the new one I also see that some lines are missing.

TD-er, 2018-06-20:
OK, that's something to look into then.
We can start by making the buffer usage of the log buffer a bit more efficient

TD-er, 2018-06-20:
By the way, the number of lines for this buffer differs between versions and is based a little on dev/normal builds, since there has to be some free memory left :)

s0170071, 2018-06-20:
OK, I cant't tell if the web viewer drops those lines too because I am using serial for the MT861. I was, however, never missing lines from serial.
And yes, I use only one viewer.
Just as a precaution (and to avoid any heap out of scope issues) I use a global string and pass it to the addLog function.

TD-er, 2018-06-20:
You could increase the number of log lines in the buffer to see if this makes a difference.
https://github.com/letscontrolit/ESPEasy/blob/96367a9422b1c83ff38f47cbda9c12dee23f95af/src/ESPEasy-Globals.h#L989-L998

s0170071, 2018-06-20:
A log from the browser and the json data it received.
The window displays no line MT681: 37

```
 
53070: MT681: 35 e2s:1076 s2s:1560 s2e:464 e2e:1540
53479: connect: 192.168.1.41 OK, sending read request: 0 1 0 0 0 6 3 4 119 53 0 4
53671: reading bytes: 0 1 0 0 0 11 3 4 8 > 0 0 0 0 0 0 178 152 value: 45720.00
54630: MT681: 36 e2s:1095 s2s:1559 s2e:465 e2e:1560
57730: MT681: 38 e2s:1096 s2s:1560 s2e:464 e2e:1560
58294: connect: 192.168.1.41 OK, sending read request: 0 1 0 0 0 6 3 4 119 53 0 4
58471: reading bytes: 0 1 0 0 0 11 3 4 8 > 0 0 0 0 0 0 178 152 value: 45720.00
59290: MT681: 39 e2s:1095 s2s:1559 s2e:464 e2e:1560
 
```

json data for lines 36,37,38 shows: 

{"Log": {"Entries": [{"timestamp":54630,
"text":"MT681: 36 e2s:1095 s2s:1559 s2e:465 e2e:1560",
"level":3}],
"TTL":1000,
"timeHalfBuffer":1000,
"nrEntries":1,
"SettingsWebLogLevel":4,
"logTimeSpan":0
}}

{"Log": {"Entries": [{"timestamp":55879,
"text":"connect: 192.168.1.41 OK, sending read request: 0 1 0 0 0 6 3 4 120 55 0 2 ",
"level":3},
{"timestamp":56071,
"text":"reading bytes: 0 1 0 0 0 7 3 4 4 > 128 0 0 0 value: -2147483648.00",
"level":3},
{"timestamp":56170,
"text":"MT681: 37 e2s:1075 s2s:1540 s2e:464 e2e:1540",
"level":3}],
"TTL":727,
"timeHalfBuffer":727,
"nrEntries":3,
"SettingsWebLogLevel":4,
"logTimeSpan":291
}}

{"Log": {"Entries": [],
"TTL":1000,
"timeHalfBuffer":1000,
"nrEntries":0,
"SettingsWebLogLevel":4,
"logTimeSpan":0
}}

{"Log": {"Entries": [{"timestamp":57730,
"text":"MT681: 38 e2s:1096 s2s:1560 s2e:464 e2e:1560",
"level":3}],
"TTL":1000,
"timeHalfBuffer":1000,
"nrEntries":1,
"SettingsWebLogLevel":4,
"logTimeSpan":0
}}

What is this third json block ? Empty ?

TD-er, 2018-06-20:
So the problem looks to be JavaScript related and not the buffer on the ESP node.
Indeed the 3rd block looks to be empty, meaning no data was stored in the buffer between reads from the JavaScript.

It is strange, since the JSON was requested.
Maybe some async issue in JavaScript?

Grovkillen, 2018-08-04:
I'm planning on doing some testing and fixing this next week. Is it still an issue @s0170071 ? @TD-er did some tweaking I see. Does this help or not?

TD-er, 2019-10-10:
Is this still an issue that needs to be dealt with, or have other optimizations in the code make it fast enough not to miss any log item anymore?

closed 2019-10-27 by TD-er