letscontrolit/ESPEasy #1525: MQTT failure if second AP switched on

s0170071, 2018-06-22:
If I switch on my second FritzBox (uses the same SSID and PW like the first one) MQTT fails. The webpage is still reachable and the ESP seems to operate normal. Apart from MQTT that is.

Info page shows: 
Build
20102  - Mega
Libraries
ESP82xx Core ea4720b0, NONOS SDK 2.2.1(cfd48f3), LWIP: 2.0.3
GIT version
(custom)
Plugins
74 [Normal] [Testing]
Build time
Jun 20 2018 22:58:48
Binary filename
ThisIsTheDummyPlaceHolderForTheBinaryFilename64ByteLongFilenames

Unit | 17
-- | --
Uptime | 1 days 10 hours 2 minutes
Load | 15% (LC=11782)
Free Mem | 13416 (6872 - addLog)
Boot | Cold boot (0)
Reset Reason | External System
Network ❔
IP config | DHCP
IP / subnet | 0.0.0.0 / 0.0.0.0
GW | 0.0.0.0
Client IP | 192.168.1.65
DNS | 192.168.1.1 / 0.0.0.0
Allowed IP Range | 0.0.0.0 - 0.0.0.0
Serial Port available: | 0 (0 , 203)
**STA MAC | 2C:3A:E8:07:02:2F**
AP MAC | 2E:3A:E8:07:02:2F
SSID | MNET (5C:49:79:28:0C:6F)
Channel | 6
Connected | 7h50m
Last Disconnect Reason | (200) Beacon timeout
Number reconnects | 1
Firmware
Build | 20102  - Mega
Libraries | ESP82xx Core ea4720b0, NONOS SDK 2.2.1(cfd48f3), LWIP: 2.0.3
GIT version | (custom)
Plugins | 74 [Normal] [Testing]
Build Md5 | 4d44355f4d44355f4d44355f4d44355f
Md5 check | fail !
Build time | Jun 20 2018 22:58:48
Binary filename | ThisIsTheDummyPlaceHolderForTheBinaryFilename64ByteLongFilenames
ESP board
ESP Chip ID | 459311 (0x7022F)
ESP Chip Freq: | 80 MHz
Storage
Flash Chip ID | Vendor: 0xC8 Device: 0x4016
Flash Chip Real Size: | 4096 kB
Flash IDE Size: | 4096 kB
Flash IDE speed: | 40 MHz
Flash IDE mode: | DIO
Flash Writes | 0 daily / 1 boot
Sketch Size | 761 kB (2308 kB free)

Mosquitto log shows: 
1529611225: New connection from 192.168.1.12 on port 1883.
1529611225: New client connected from 192.168.1.12 as ESPClient_5C:CF:7F:39:55:D3 (c1, k10, u'').
1529611342: Client ESPClient_5C:CF:7F:39:B7:B0 has exceeded timeout, disconnecting.
1529611342: Socket error on client ESPClient_5C:CF:7F:39:B7:B0, disconnecting.
1529611342: New connection from 192.168.1.24 on port 1883.
1529611342: New client connected from 192.168.1.24 as ESPClient_5C:CF:7F:39:B7:B0 (c1, k10, u'').
**1529623938: Socket error on client ESPClient_2C:3A:E8:07:02:2F, disconnecting.**
1529646284: Client ESPClient_5C:CF:7F:39:55:D3 has exceeded timeout, disconnecting.
1529646284: Socket error on client ESPClient_5C:CF:7F:39:55:D3, disconnecting.
1529646303: New connection from 192.168.1.12 on port 1883.

ESP (weblog, debug more) shows nothing other than my MT681 was not called for >500ms (should be 20ms), irregular intervals, several seconds apart.

The connection does not recover by itself if I unplug the second FritzBox.
a reboot does fix the issue. And I now see UDP messages in the weblog. These were not there when the MQTT was not working.

TD-er, 2018-06-22:
Hmm, that's a bit strange.

I was thinking about adding some kind of "network watchdog" to see if any traffic is possible, or else reboot.
But judging from this report, it is also possible to have network traffic and still not being able to make connections.
Ofcourse I could look at IP-address and if it is still not set, try to restart the WiFi.

s0170071, 2018-06-22:
Yea, there was traffic. Just not MQTT. Thing is, that there is little debug info on that. I was only seeing only output from my own plugins. (got to admit I removed the webserver logging as it is annoying when you use the weblog.)

TD-er, 2018-06-22:
The MQTT connection manager does test for network before connecting.
So maybe we should make this test a bit more elaborate.

s0170071, 2018-06-22:
I did not reconnect when I removed the second AP. Does the connection manager retry or can it get into a state where it won't retry ?

TD-er, 2018-06-22:
I will not say it cannot reach some undefined state ;)
It should try to reconnect, but this looks like some kind of bug, so I guess anything is possible.
The IP-address shown on the info page is information requested from the core library and whether or not to reconnect is decided based on another state from the same library.
So if these are out of sync, then it is hard to say what should happen.

s0170071, 2018-07-09:
Quickfix for openhab users: 
As the ESP is still available via http web interface (despite not sending MQTT) you can use this to reboot it:

1. use the OpenHab "expire" addon to set all overdue MQTT readings to "UNDEF". 

2. create a rule which triggers on a variable (e.g. temperature) change to "UNDEF" and send a "http://192.168.1.x/?cmd=reboot" http get request to the ESP. 

This resets the ESP and resolves the issue. Happens about once a week.

TD-er, 2019-10-27:
Is this still an issue?

TD-er, 2022-04-01:
Should be fixed by these PRs: 
- https://github.com/letscontrolit/ESPEasy/pull/3987
- https://github.com/letscontrolit/ESPEasy/pull/3983

Please let me know if it is indeed fixed now... finally

closed 2022-04-01 by TD-er