letscontrolit/ESPEasy #1526: do not log webserver when using the web log

s0170071, 2018-06-22:
if you use the web log you see a lot of output from the web server serving the web log.
I think there should be a specific option to mute those log entries that are caused by the web log. 

Should be doable as the webserver "knows" which page was requested. If it is the log, logging is inhibited by using a temporary no-log flag...

TD-er, 2018-06-22:
This webserver logs can be removed (put behind a flag). I now know how the flash strings are processed by the chunked transfers.

And indeed, there will be some filtering added in the (near?) future.
I still have to look into this JavaScript sometimes missing a line issue. And now that @Grovkillen is on holiday, I have to do it myself (don't know JavaScript yet), so maybe adding a few features myself could be some good exercise ;)

s0170071, 2018-06-22:
Well, the webserver logs in general are ok, but not those that a caused by the web log. Those are a bit like a dog trying to catch its tail...

TD-er, 2018-06-22:
And in the end, they will result in an error... I know from experience.

I will remove them.

tonhuisman, 2022-04-18:
This seems to have been fixed, so can probably be closed.

closed 2022-04-18 by TD-er