letscontrolit/ESPEasy #1527: ESPEasy (Sonoff 4ch) - Won't connect to network

Sascha-Zinn, 2018-06-22:
I Just flashed my Sonoff 4 channel with the latest ESPEasy firmware. i connected to the access point created by the device, selected the WiFi network to connect to, keyed in the password, it counted down from 20 and nothing. It gives me an option to return to setup.

I have re-flashed, i have tried other networks, but it just isn't connecting.

Any suggestions would be appreciated.

TD-er, 2018-06-22:
As a work-around, you can also set the network config via a serial monitor using the [ESPeasy commands](https://www.letscontrolit.com/wiki/index.php/ESPEasy_Command_Reference).
Or using the tools made by @Grovkillen . See his [signature links on the forum](https://www.letscontrolit.com/forum/memberlist.php?mode=viewprofile&u=2112).

Sascha-Zinn, 2018-06-22:
I tried this and still not working.

I logged into the AP created by the Sonoff switch, checked the log and this is what it says:

1395: WIFI : AP Mode ssid will be ESP_Easy_0 with address 192.168.4.1
1402: WD   : Uptime 0 ConnectFailures 0 FreeMem 21696
13976: AP Mode: Client connected: MACaDDRESS Connected devices: 1
13979: Webserver: start
24138: WIFI : Set WiFi to AP
31402: WD   : Uptime 1 ConnectFailures 0 FreeMem 17472
57697: FILE : Saved config.dat
58151: FILE : Saved security.dat
58212: WIFI : Set WiFi to AP+STA
58244: WIFI : Connecting ZeFunHaus attempt #0
61740: WD   : Uptime 1 ConnectFailures 0 FreeMem 16352
61763: WIFI : Connected! AP: ZeFunHaus (MACaDDRESS) Ch: 8 Duration: 3518 ms
63482: AP Mode: Client disconnected: MACaDDRESS Connected devices: 0
63513: AP Mode: Client connected: MACaDDRESS Connected devices: 1
91740: WD   : Uptime 2 ConnectFailures 0 FreeMem 15704
121740: WD   : Uptime 2 ConnectFailures 0 FreeMem 15704

TD-er, 2018-06-22:
What version of the firmware did you use? A pre-built one of self-built?

Sascha-Zinn, 2018-06-22:
Pre-Built.

TD-er, 2018-06-22:
I had to change a few things in the board definitions for the ESP8285 to be able to compile it again with the latest versions of PlatformIO.
This newly introduced setting was about the max upload size, which is new in PlatformIO.
Maybe it also affects the position (and operations) of the SPIFF, where the settings are stored?

Could you try a version of a few days ago, just to be sure?

Sascha-Zinn, 2018-06-22:
Flashed to version:
ESP_Easy_mega-20180524_normal_ESP8285_1024
And still the same issue.

464: WIFI : Could not connect to AP!
467: WIFI : Set WiFi to AP
1388: WIFI : AP Mode ssid will be ESP_Easy_0 with address 192.168.4.1
1394: WD : Uptime 0 ConnectFailures 0 FreeMem 22216
8753: AP Mode: Client connected: MAC Connected devices: 1
8757: Webserver: start
18639: WIFI : Set WiFi to AP
28474: WIFI : Credentials Changed, retry connection. SSID: ZeFunHaus
28502: WIFI : Set WiFi to AP+STA
28534: WIFI : Connecting ZeFunHaus attempt #0
32026: WD : Uptime 1 ConnectFailures 0 FreeMem 16568
32065: WIFI : Connected! AP: ZeFunHaus (MAC) Ch: 8 Duration: 3530 ms
33262: AP Mode: Client disconnected: MAC Connected devices: 0
33293: AP Mode: Client connected: MAC Connected devices: 1
62027: WD : Uptime 1 ConnectFailures 0 FreeMem 16352

Sascha-Zinn, 2018-06-22:
I was setting up my switch according to a youtube video and rolled back to the version they were using : 
ESP_Easy_v2.0-20180220_normal_ESP8285_1024

And its working fine now 👍

TD-er, 2018-06-23:
Hmm, that's an old one. I hope there is some newer version that will work on the 4CH units.

clumsy-stefan, 2018-06-23:
I use a latest ESPEasy GIT (self compiled) version with a board-definitioon of 8266 1M/128K (not8285) and is working fine on my  two 4CH units (one from Sonoff and a no-name china one)... just as information, if it helps...

jornvdp, 2018-07-03:
It  looks like I'm facing the same problem at the moment.
Flashed my 4ch. Sonoff(China Clone I think, about 25 euro) with the latest version(ESPEasy_mega-20180625 normal ESP8285 1024) and the Esp_0 device appears.
When I connect to it, I can select my ssid(the strongest one, I have 3 ap's with the same ssid'd) but after 20 seconds I can only go back to setup and try again.
It looks like it's connected to the access point(I can see and Esp_0 device is connected), but don't gets an ip address.
I have also tried older versions of firmware(even ESPEasy_v2.0-20180220) but that has the same behaviour,
An ESP-12E(also from China) connects without any problems.

I've searched in the logging from my dns server and it seems that the device gets an ip normally, but the Sonoff don't gets back with an ip result.

Logging Sonoff serial(I tried 2 times in this log):

`INIT : Booting version: mega-20180625 (ESP82xx Core 2_4_1, NONOS SDK 2.2.1(cfd48f3), LWIP: 2.0.3)
102 : INIT : Cold Boot - Restart Reason: Power on
106 : FS   : Mounting...
112 : FS   : Mount successful, used 75802 bytes of 113201
427 : CRC  : program checksum       ...OK
434 : CRC  : SecuritySettings CRC   ...OK
435 : CRC  : binary has changed since last save of Settings
454 : INIT : Free RAM:24168
454 : INIT : I2C
454 : INIT : SPI not enabled
469 : INFO : Plugins: 48 [Normal] (ESP82xx Core 2_4_1, NONOS SDK 2.2.1(cfd48f3), LWIP: 2.0.3)
470 : WIFI : No valid wifi settings
470 : WIFI : Could not connect to AP!
473 : WIFI : Set WiFi to AP
1396 : WIFI : AP Mode ssid will be ESP_Easy_0 with address 192.168.4.1
1403 : WD   : Uptime 0 ConnectFailures 0 FreeMem 21888
23741 : AP Mode: Client connected: A4:34:D9:0A:51:B3 Connected devices: 1
23745 : Webserver: start
27423 : WIFI : Set WiFi to AP
31403 : WD   : Uptime 1 ConnectFailures 0 FreeMem 15400
39571 : WIFI : Credentials Changed, retry connection. SSID: JornIoT
39603 : WIFI : Set WiFi to AP+STA
39636 : WIFI : Connecting JornIoT attempt #0
43150 : WIFI : Connected! AP: JornIoT (44:D9:E7:FA:60:6A) Ch: 11 Duration: 3513 ms
44930 : AP Mode: Client disconnected: A4:34:D9:0A:51:B3 Connected devices: 0
44970 : AP Mode: Client connected: A4:34:D9:0A:51:B3 Connected devices: 1
61403 : WD   : Uptime 1 ConnectFailures 0 FreeMem 13536
72511 : WIFI : Credentials Changed, retry connection. SSID: JornIoT
72541 : WIFI : Connecting JornIoT attempt #0
91403 : WD   : Uptime 2 ConnectFailures 0 FreeMem 13992
98396 : AP Mode: Client disconnected: A4:34:D9:0A:51:B3 Connected devices: 0
121403 : WD   : Uptime 2 ConnectFailures 0 FreeMem 15000
151403 : WD   : Uptime 3 ConnectFailures 0 FreeMem 15000
181403 : WD   : Uptime 3 ConnectFailures 0 FreeMem 15000
211403 : WD   : Uptime 4 ConnectFailures 0 FreeMem 15000
241403 : WD   : Uptime 4 ConnectFailures 0 FreeMem 15000
271403 : WD   : Uptime 5 ConnectFailures 0 FreeMem 15000
301403 : WD   : Uptime 5 ConnectFailures 0 FreeMem 15000
331403 : WD   : Uptime 6 ConnectFailures 0 FreeMem 14832
361403 : WD   : Uptime 6 ConnectFailures 0 FreeMem 14832
391403 : WD   : Uptime 7 ConnectFailures 0 FreeMem 14832
421403 : WD   : Uptime 7 ConnectFailures 0 FreeMem 14832`

And logging from my dns server(dsnmasq)
`Jul  3 19:14:17 router dnsmasq-dhcp[17057]: DHCPDISCOVER(eth1.99) 60:01:94:8e:2c:2c
Jul  3 19:14:17 router dnsmasq-dhcp[17057]: DHCPOFFER(eth1.99) 192.168.99.14 60:01:94:8e:2c:2c
Jul  3 19:15:19 router dnsmasq-dhcp[17057]: DHCPDISCOVER(eth1.99) 60:01:94:8e:2c:2c
Jul  3 19:15:19 router dnsmasq-dhcp[17057]: DHCPOFFER(eth1.99) 192.168.99.14 60:01:94:8e:2c:2c
Jul  3 19:16:19 router dnsmasq-dhcp[17057]: DHCPDISCOVER(eth1.99) 60:01:94:8e:2c:2c
Jul  3 19:16:19 router dnsmasq-dhcp[17057]: DHCPOFFER(eth1.99) 192.168.99.14 60:01:94:8e:2c:2c`

TD-er, 2018-07-03:
Can you also try to set the IP config using the [flash tool](https://github.com/Grovkillen/ESP_Easy_Flasher) made by @Grovkillen ?

That one can also help setup the IP config right after flashing.

Grovkillen, 2018-07-04:
> Also no success, it looks like the esp is even ignoring the given ip(i
> tried with dhcp and also with fixed ip)
>
> ######POST FLASH######
> [2018-07-04 18:13:52] Sending SERIAL: "Name Sonoff-4ch-Garage"
> [2018-07-04 18:13:54] Sending SERIAL: "WifiSSID JornIoT"
> [2018-07-04 18:13:56] Sending SERIAL: "WifiKey *******"
> [2018-07-04 18:13:58] Sending SERIAL: "IP 192.168.99.99"
> [2018-07-04 18:14:00] Sending SERIAL: "Save"
> [2018-07-04 18:14:02] Sending SERIAL: "WifiConnect"
> [2018-07-04 18:14:02] Serial programming done.
> [2018-07-04 18:14:24] Unit IP is 0.0.0.0.
> [2018-07-04 18:14:24] Will test HTTP connection now...
> [2018-07-04 18:14:24] Pinging 0.0.0.0.
> [2018-07-04 18:14:25] ERROR: 0.0.0.0 is unresponsive.
> ############################
> [2018-07-04 18:14:26] JOB COMPLETED, over and out!

If you reboot the unit and then click on serial monitor you'll see that the
unit do in fact get an IP. My tool need to be updated...

TD-er, 2021-08-12:
This is no longer an issue, as far as I know.
So will close it now.

closed 2021-08-12 by TD-er