letscontrolit/ESPEasy #1530: OpenhabMQTT not reconnecting

henfri, 2018-06-23:
### Summarize of the problem/feature request

I am using ESPEasy to monitor the temperature of a fridge ( :-) ). The Wifi coverage is not very good, but ok. A ping works in 2/3 of the time.
The temperature is transmitted via OpenhabMQTT reliably for a time (hours). Then it stops.
In the log I see "HTTP : connection failed" regularly.
When stopping and starting the OpenhabMQTT  Controller ( [ ] enable -> save -> [x] enable) it works again.

### Expected behavior

The mqtt controller should re-initialize after a number of unsuccessful connects.

### Actual behavior

if the connection failed once, it will not reconnect

### Steps to reproduce

1. Use OpenhabMQTT on a bad wifi connection
2. Monitor a Sensor for a while
3. This will fail after a while and you will see "HTTP : connection failed" in the logs

After reboot the problem is fixed again for a while. But even disable/enable of MQTT solves it (temporarily)

### System configuration

Hardware:
NodeMCU

ESP Easy version:  20102 - Mega

Grovkillen, 2018-06-23:
If you did a reboot on regular basis I think it would reset the missed
requests counter that will trigger if too many connection attempts is
reached. @TD-er correct me if I'm wrong.

henfri, 2018-06-23:
Hello,
thanks for your reply!

so this ist an intended behavior?
Is this counter reset automatically after a successful attempt?

Can I increase the threshold?

How can I schedule a reboot on regular basis as workaroud?

Greetings,
Hendrik

Grovkillen, 2018-06-23:
In a "normal" scenario the intent is to not hammer the controller unit with requests if something is wrong. But your scenario is a bit different I guess.

henfri, 2018-06-23:
So, what would be your recommendation to workaround this?

But I am not sure that my scenario is so unusual. I think that a automatic reconnect (maybe with increasing delay) would be worth a consideration. 

MQTT is made for unreliable connections. The ESP will not overwhelm the controller if trying every 10/100/1000/10000 seconds.

Greetings,
Hendrik

TD-er, 2018-06-23:
It may be related to this issue: https://github.com/letscontrolit/ESPEasy/issues/1525
Wifi stability is still an ongoing issue, which I have to address very soon.
Also the detection of whether or not a connection is active may prevent MQTT connection manager to attempt a reconnect.

fmuntean, 2018-06-28:
I am having the same issue here. The WiFi connection has no problem and I can access the web page and ping the device with no issue.
It also seems that the MQTT#Disconnected event is not fired thus I can't use a rule to reboot the device. 
I have Build | 20100  - Mega (core 2_4_1)

uzi18, 2018-07-02:
@TD-er I have also problems with this Controller, it sends data irregularly, will try investigate more.

small fix for this controller PR#1546

TD-er, 2019-10-27:
Is this still an issue?

TD-er, 2019-11-04:
This one is probably not an issue anymore, so I will close it.
Please let me know if it still is an issue.

closed 2019-11-04 by TD-er