letscontrolit/ESPEasy #1534: Improper GET(s) on HTTP Controller

ilgrank, 2018-06-24:
### Summarize of the problem/feature request

Hello everyone

I noticed that the Generic HTTP controller isn't behaving as expected.
When making a GET versus the target controller,the connection is RST before receiving the 200 from the controller.. looking at a wireshark capture: (**.249 is the ESPEasy, .101 is the Controller**):

### Actual behavior
```
192.168.0.249	192.168.0.101	HTTP	174	GET /ESPtest.php?name=ESP_Easy&task=Button&valuename=Switch&value=1 HTTP/1.1 
192.168.0.101	192.168.0.249	TCP	54	80 → 49392 [FIN, ACK] Seq=1 Ack=121 Win=65392 Len=0
**192.168.0.249	192.168.0.101	TCP	60	49392 → 80 [RST, ACK] Seq=121 Ack=2 Win=2144 Len=0**
```
the ESPEasy RST immediately, and the server does not send 200(ok)

### Expected behavior
making the same GET from a simple browser (again, 249 is espeasy, 110 is a computer with Firefox:

```
192.168.0.110	192.168.0.101	HTTP	508	GET /ESPtest.php?name=ESP_Easy&task=Uptime&valuename=Upt&value=75.00 HTTP/1.1 
192.168.0.101	192.168.0.110	TCP	344	80 → 13657 [PSH, ACK] Seq=1 Ack=455 Win=65536 Len=290 [TCP segment of a reassembled PDU]
192.168.0.110	192.168.0.101	TCP	60	13657 → 80 [ACK] Seq=455 Ack=291 Win=525056 Len=0
**192.168.0.101	192.168.0.110	HTTP	85	HTTP/1.1 200 OK** 

```
works as it should.
(not all webservers are sensible to this Kind of behaviour, Apache tolerates it to some extent, Abyss webserver does not at all)

I think it _could_ be related to what was exposed [here](https://github.com/letscontrolit/ESPEasy/issues/1066)

### Steps to reproduce
Configure a controller with Generic HTTP, with controller publish to eg:
`ESPtest.php?name=%sysname%&task=%tskname%&valuename=%valname%&value=%value%`

sample ESPTest.php:
```
<?php
$log="ESPtest.log";
$date = date('l jS \of F Y h:i:s A');
$data=str_replace('=', ':', http_build_query($_GET, null, ','));
error_log($date." ".$data."\n", 3, $log);
?>
```

use Abyss Web server with PHP package 5.x to experience the full issue:
[Abyss Web Server Free, latest](https://aprelium.com/abyssws/download.php)

Configure a simple Device, such as System Uptime, to send data every 60 seconds.

calling this url from a browser
<IP of controller>/ESPtest.php?name=ESP_Easy&task=Uptime&valuename=Upt&value=75.00
will trigger the log
the ESPEasy won't.
Using wireshark on the Controller reveals the data above (premature RST)

### System configuration
```
Build | 20102 - Mega
Libraries | ESP82xx Core 2_4_1, NONOS SDK 2.2.1(cfd48f3), LWIP: 2.0.3
GIT version | mega-20180624
Plugins | 48 [Normal]
Build Md5 | a1b9e2f28f973d7d0564bfdd35792e5
Md5 check | passed.
Build time | Jun 24 2018 02:15:14
Binary filename | ESP_Easy_mega-20180624_normal_ESP8266_1024.bin

```
Hardware in the example is a Sonoff S20, but I've reproduced it with an ESP-12 too.

### Rules or log data
Rules: none.

Log data:
```
6423111: SYS : 107.00
6423113: EVENT: Uptime#Upt=107.00
6423708: WD : Uptime 107 ConnectFailures 0 FreeMem 18808
6453111: SYS : 107.00
6453113: EVENT: Uptime#Upt=107.00
6453708: WD : Uptime 108 ConnectFailures 0 FreeMem 18808
```

ilgrank, 2018-07-17:
Is there something I can do? 
The problem is still here even in the latest build, and it is easy to reproduce.. yet not even a comment :/

Grovkillen, 2018-07-17:
It's holidays here in Sweden and my dutch bestest friend is soon to join. We will gain momentum this autumn but right now we're either investing time in our family or (payed) job. Only evenings/nights/weekends are currently dedicated to ESP Easy. This will hopefully change within a couple of month if we could get some funds to do this as a part time job. Nothing is decided yet but we hope to be able to spend time on this like daytime job... fingers crossed!

TD-er, 2018-07-17:
The last few weeks (months?) I was very busy with personal issues (regarding our houses, not relational issues) and like @Grovkillen said, we only have a few hours a day to spend on the project.

The reason why I did not spend time on this one yet is that it will take some time to setup a test and look into the code.
It may be the bug is related to the core libraries and then it will be taking even more time to fix or create a work-around.

Also we're working very hard in getting ESPeasy stable and I don't expect this issue to be causing stability issues.  So it is at this moment a matter of priorities.
And believe me, I would really like to fix all issues, but there is only a limited amount of time in a week.

So you could help in creating a very easy to setup test.

Grovkillen, 2018-07-18:
Donation received, thank you! :) I wish we had that on a regular basis instead of once a month :P

ilgrank, 2018-07-18:
Do you have a Patreon account too?

Grovkillen, 2018-07-18:
I'm planning on getting one, will for sure do it soon. Would you like it, will others follow you think?

ilgrank, 2018-07-18:
I do support *lots* of devs  (I do nearly 150 donations/year, mostly ranging from $5 to $10), so I prefer to make small donations, yearly.. but I see potential in Patreon-style recurring support.
I was not joking about the fact that if every person who downloaded ESPEasy donated even just $1-2 you could make a full time wage out of it imho

tonhuisman, 2023-09-03:
This seems to be solved, so can be closed.

closed 2023-09-04 by TD-er