letscontrolit/ESPEasy #1535: rules with REST API

jessie00, 2018-06-26:
Hi,
Have you Idea to use REST API with rules.
I have BOSE soundtoutch, when I use curl like this on my PC I power ON / OFF the speaker:

curl -s -o /dev/null --data '
    <key state="press" sender="Gabbo">POWER</key>
    <key state="release" sender="Gabbo">POWER</key>
' \
http://192.168.0.6:8090/key

Have you a idea to send this directly to the speaker.
Today i use mqtt server with a raspberry to send command. #

TD-er, 2018-06-26:
Using rules you can send all kinds of http requests.
But this is more a question on how to use things and not a request to change the software.
So I think you're better off asking this on the [forum](https://www.letscontrolit.com/forum/viewforum.php?f=1)

Grovkillen, 2018-07-11:
I close this for now

closed 2018-07-11 by Grovkillen