letscontrolit/ESPEasy #1536: Generic HTTP Advanced crash

mickeprag, 2018-06-27:
### Summarize of the problem/feature request
When using Generic HTTP Advanced the ESP reboots

### Expected behavior
The ESP should make a HTTP call and not crash

### Actual behavior
The esp crashes and reboots. The serial output below

### Steps to reproduce
[image]

### System configuration

Hardware: NodeMCU Amica

ESP Easy version: mega-20180625 testing

### Rules or log data

```
INIT : Booting version: mega-20180625 (ESP82xx Core 2_4_1, NONOS SDK 2.2.1(cfd48f3), LWIP: 2.0.3)
89 : INIT : Warm boot #18 - Restart Reason: Exception
92 : FS   : Mounting...
118 : FS   : Mount successful, used 76053 bytes of 957314
457 : CRC  : program checksum       ...OK
488 : CRC  : SecuritySettings CRC   ...OK 
593 : INIT : Free RAM:22384
594 : INIT : SPI not enabled
601 : HX711: GPIO: SCL=0 DOUT=4
612 : INFO : Plugins: 73 [Normal] [Testing] [Development] (ESP82xx Core 2_4_1, NONOS SDK 2.2.1(cfd48f3), LWIP: 2.0.3)
612 : WIFI : Set WiFi to STA
645 : WIFI : Connecting Telldus attempt #0
654 : WD   : Uptime 0 ConnectFailures 0 FreeMem 21504
659 : HX711: Value: NO NEW VALUE
4422 : WIFI : Connected! AP: **** (XX:XX:XX:XX:XX:XX) Ch: 6 Duration: 3775 ms
4441 : WIFI : DHCP IP: 192.168.0.121 (my-sensor-0) GW: 192.168.0.1 SN: 255.255.255.0   duration: 18 ms
4444 : Webserver: start
14346 : FILE : Saved config.dat
14551 : FILE : Saved config.dat
14553 : HX711: GPIO: SCL=0 DOUT=4
30654 : WD   : Uptime 1 ConnectFailures 0 FreeMem 17456
60649 : HX711: Value: 14170.000 = 0.899
61772 : Timeout while reading input data!
61781 : WD   : Uptime 1 ConnectFailures 0 FreeMem 17456
81970 : FILE : Saved config.dat
82165 : FILE : Saved config.dat
82167 : HX711: GPIO: SCL=0 DOUT=4
91781 : WD   : Uptime 2 ConnectFailures 0 FreeMem 17504
120649 : HX711: Value: 14174.000 = 0.899

Panic /home/travis/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/core_esp8266_main.cpp:99 __yield

                                                                                                                     ctx: sys 
                                                                                                                              sp: 3fff31e0 end: 3fffffb0 offset0

>>>stack>>>
           3fff3390:  3fff6dcc 00000030 3fff78f4 40100670  
3fff33a0:  00000024 00003010 00000602 4025d5b8  
3fff33b0:  00000011 00000001 0000000a 4020cf2d  
3fff33c0:  00000030 00000001 3fff3820 4020cf74
```

mickeprag, 2018-06-27:
It works in v2.0-20180321 so something has happened since then.

TD-er, 2018-06-27:
That's about the time when I changed the way how the WiFi makes connection.
So it is probably related to not checking if there is an connection and then the request just waits and doesn't make calls to `yield()` or does do `delay()`.

This means it could always lead to crashes, but now it is easier to reproduce if this call is made right after boot.

In older versions the entire wifi connection was done in the `setup()` function, which means it could take up to 30'ish seconds to boot and show some activity.
Now it just starts the wifi connection sequence and continues to do the rest. When an event happens signaling the next step in WiFi connection, it processes the data from that step and continues with the next step.
This makes the process a lot faster, but it also means functions talking to the outside world should check for active connections or else things will fail. My first guess is that's what's happening here.

mickeprag, 2018-06-27:
Is there anything I can do to help test or pinpoint the issue, please ask.

Unfortunately I do not have the time to dig into the code but testing a patch or bisecting the correct commit or something similar, that I can do.

TD-er, 2018-06-27:
Well pinpointing the moment would be hard, since at that time, we also had an issue with bad nightly builds.
Probably due to some bug in the linking process.
When building twice, the builds are fine and that took over a month of trying to fix non-existent bugs due to not reproducible bug reports.
And that's when I changed a lot in the WiFi code.

popaserge, 2018-08-03:
Hello
I had the same crash and tried the v2.0-20180321 version as said by mickeprag.

As I tried to send 3 DS18b20 températures at the same time I configured 3 devices, named them Temp1, Temp2 and Temp3, each of them sending the result to the controller at the same frequency (delay?).

The controller Http body is a Json string:
{ "xxxx" : { "result" : [ 
{"Channel" : "%vname1%", "Unit" : "Temperature", "Float" : "1" , "Decimalmode": "2" , "Value" : "%val1%"};
{"Channel" : "%vname2%", "Unit" : "Temperature", "Float" : "1" , "Decimalmode": "2" , "Value" : "%val2%"};
{"Channel" : "%vname3%", "Unit" : "Temperature", "Float" : "1" , "Decimalmode": "2" , "Value" : "%val3%"}]}}

It seems that the controller is triggered by each of the device sending the data and on the serial output it can be seen the whole json string alternatively with 1 of 3 temperature values but never together.

HTTP before parsing:
{ "xxxx" : { "result" : [
{"Channel" : "%vname1%", "Unit" : "Temperature", "Float" : "1" , "Decimalmode": "2" , "Value" : "%val1%"};
{"Channel" : "%vname2%", "Unit" : "Temperature", "Float" : "1" , "Decimalmode": "2" , "Value" : "%val2%"};
{"Channel" : "%vname3%", "Unit" : "Temperature", "Float" : "1" , "Decimalmode": "2" , "Value" : "%val3%"}]}}
HTTP after parsing:
{ "xxxx" : { "result" : [
{"Channel" : "%vname1%", "Unit" : "Temperature", "Float" : "1" , "Decimalmode": "2" , "Value" : "%val1%"};
{"Channel" : "%vname2%", "Unit" : "Temperature", "Float" : "1" , "Decimalmode": "2" , "Value" : "%val2%"};
{"Channel" : "%vname3%", "Unit" : "Temperature", "Float" : "1" , "Decimalmode": "2" , "Value" : "%val3%"}]}}
HTTP after replacements:
{ "xxxx" : { "result" : [
{"Channel" : "Temp1", "Unit" : "Temperature", "Float" : "1" , "Decimalmode": "2" , "Value" : "27.87"};
{"Channel" : "", "Unit" : "Temperature", "Float" : "1" , "Decimalmode": "2" , "Value" : "0.00"};
{"Channel" : "", "Unit" : "Temperature", "Float" : "1" , "Decimalmode": "2" , "Value" : "0.00"}]}}

Then the string after replacement changes
{"Channel" : "Temp2", "Unit" : "Temperature", "Float" : "1" , "Decimalmode": "2" , "Value" : "27.31"};
{"Channel" : "", "Unit" : "Temperature", "Float" : "1" , "Decimalmode": "2" , "Value" : "0.00"};
{"Channel" : "", "Unit" : "Temperature", "Float" : "1" , "Decimalmode": "2" , "Value" : "0.00"}]}}
and
{"Channel" : "Temp3", "Unit" : "Temperature", "Float" : "1" , "Decimalmode": "2" , "Value" : "27.44"};
{"Channel" : "", "Unit" : "Temperature", "Float" : "1" , "Decimalmode": "2" , "Value" : "0.00"};
{"Channel" : "", "Unit" : "Temperature", "Float" : "1" , "Decimalmode": "2" , "Value" : "0.00"}]}}

Conclusion : each time a new connexion to the server is made and the placeholders for the device names and values are not very usefull.
Would it not be more convenient if the controller was set with a working frequency to fetch the last device values stored somewhere by the devices and then make only 1 connexion and send the data.
Could that be a reason for the crash on the last release.
I am not able to code something and to check on the last version because of the crash but I can ckeck the  result of any modification. (hardware : Wemos D1 mini)

popaserge, 2018-08-07:
I Tried with only one line in http body with reference to only the first device placeholder %vname1% and %val1%

{ "xxxx" : { "result" :  
{"Channel" : "%vname1%", "Unit" : "Temperature", "Float" : "1" , "Decimalmode": "2" , "Value" : "%val1%"}}}

When I activate 2 first devices, in the serial output the 2 placehoders are alternatively replaced by device 1 and device 2 names and values and when I activate only device 3 the replacement takes device 3 name and value.
If I replace the placeholders in the single line body by %vname2% and %val2% nothing is replaced.
Seems that only placeholder of device 1 works and there is a bug in the loops in the code.

Hope this helps. If change is made would I by able to try because of crash on the last releases?

Release ESPEasy_mega-20180804 allready crashes with different type of crash messages or reset.
Hereafter different messages if it helps
>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>
Message 1
15806 : HTTP : connecting to external server

Panic /home/travis/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/core_esp8266_main.cpp:99 __yield

                                                                                                                     ctx: sys
                                                                                                                              sp: 3fff3630 end: 3fffffb0 offset: 01b0

>>>stack>>>
           3fff37e0:  00000003 00000028 3fff2844 40261b23
3fff37f0:  3ffe939c 3fff8124 3fff2844 402627ac
3fff3800:  3fff814b 3fff814c 0000000a 40219ea9........

>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>
Message 2

3fff6860:  46047c00 0000022c 0000002a 00000006
3fff6870:  0363036
 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
v614f7c32
~ld
   ▒U119 :

INIT : Booting version: mega-20180804 (ESP82xx Core 2_4_1, NONOS SDK 2.2.1(cfd48f3), LWIP: 2.0.3)
120 : INIT : Warm boot #241 - Restart Reason: **Software Watchdog**
125 : FS   : Mounting...

>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>
Message 3

pm open,type:2 0
                15767 : DS   : Temperature: 32.31 (28-c6-f0-ac-3-0-0-ca)
15769 : Read settings: ControllerSettings index: 0

 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
v614f7c32
~ld
   ▒U119 :

INIT : Booting version: mega-20180804 (ESP82xx Core 2_4_1, NONOS SDK 2.2.1(cfd48f3), LWIP: 2.0.3)
120 : INIT : Warm boot #240 - Restart Reason: **Hardware Watchdog**
125 : FS   : Mounting...

TD-er, 2018-08-07:
@popaserge:
I have been working on something that may cause this crash for the last few days.
I will make a test build now. Could you please test it to see if it also fixes this issue?
See discussion at this issue: https://github.com/letscontrolit/ESPEasy/issues/1625

TD-er, 2018-08-07:
Build has finished:
[Dropbox link](https://www.dropbox.com/s/nxats6wfgdasj6g/ESPEasy_mega-20180804-11-g7783cf4.zip?dl=0)

popaserge, 2018-08-07:
Cycle time for the devices : 30s to have time to see what happens
ESP_Easy_mega-20180804-11-g7783cf4_dev_ESP8266_4096.bin crashes every 30 s each time it has to connect with same error message as before.
Other versions happen to crash but not often I have to check better for a good report.
In any cases no improvment in placeholders work : the same as before and following messages
When replacing with Temp1
16330 : No sensor value for TaskIndex: 0 varnumber: 1
16330 : No sensor value for TaskIndex: 0 varnumber: 2
When replacing with Temp2
17908 : No sensor value for TaskIndex: 1 varnumber: 1
17909 : No sensor value for TaskIndex: 1 varnumber: 2
When replacing with Temp2
18510 : No sensor value for TaskIndex: 2 varnumber: 1
18510 : No sensor value for TaskIndex: 2 varnumber: 2

popaserge, 2018-08-09:
Hello T-Der 
Did not have time to understand and report the differences between different version got from Dropbox.
Version referenced in the web pages with Binary File name ESP_Easy_mega-20180522-26-ga5f44a1_dev_ESP8266_4096.bin has run 20 hours  and I did not observe a crash

Checked new  Release mega-20180809 just now
It crashes immédiatly with following messages
**Message 1**
5308 : Webserver: start
5309 : WIFI  : Arduino wifi status: WL_CONNECTED ESPeasy internal wifi status: ESPEASY_WIFI_SERVICES_INITIALIZED
5440 : DS   : Temperature: 25.19 (28-c6-f0-ac-3-0-0-ca)
5442 : Read settings: ControllerSettings index: 0

Panic /home/travis/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/core_esp8266_main.cpp:99 __yield

                                                                                                                     ctx: sys
                                                                                                                              sp: 3fff3640 end: 3fffffb0 offset: 01b0

**Message 2**
5310 : Webserver: start
5310 : WIFI  : Arduino wifi status: WL_CONNECTED ESPeasy internal wifi status: ESPEASY_WIFI_SERVICES_INITIALIZED
5441 : DS   : Temperature: 25.19 (28-c6-f0-ac-3-0-0-ca)
5443 : Read settings: ControllerSettings index: 0
Fatal exception 28(LoadProhibitedCause):
                                        epc1=0x402616e2, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000004, depc=0x00000000

Exception (28):
epc1=0x402616e2 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000004 depc=0x00000000

ctx: sys
         sp: 3fff3550 end: 3fffffb0 offset: 01a0

>>>stack>>>
           3fff36f0:  0000ff00 00ff0000 ff000000 00000000
3fff3700:  3fff373c 3fff806c 0000001d 3fff3880

**Other information:**

I observed twice but could not note on which version the well known panic message but with reference to core_esp8266_main.cpp:130 __yield

TD-er, 2018-08-09:
> Version referenced in the web pages with Binary File name ESP_Easy_mega-20180522-26-ga5f44a1_dev_ESP8266_4096.bin has run 20 hours and I did not observe a crash

That's a version of almost 3 months ago.

I am currently looking into the delays used by (all) controllers, so I will also look at the HTTP controllers to see what's happening here.

popaserge, 2018-08-18:
Checked the new release Release mega-20180818
Crash immediately with the new following message version:
>>>>>>>>>>>>>>>>>>
7380 : HTTP : connecting to ........

Panic /home/travis/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/core_esp8266_main.cpp:103 __yield
>>>>>>>>>>>>>>>>>

For the only version not crashing that I know (ESP_Easy_mega-20180522-26-ga5f44a1_dev_ESP8266_4096.bin) the Generic HTTP controller works with the server I use but the Generic HTTP Advanced controller does not work.
I saw  in the serial output the following concerning the http body of the Generic HTTP Advanced controller.
>>>>>>>>>>>>>>>>
Content-type: application/json ;charset=utf-8
Content-Length: 135
content={ "xxxx" : { "result" : {"Channel" : "Temp1", "Unit" : "Temperature", "Float" : "1" , "Decimalmode": "2" , "Value" : "22.81"}}}
>>>>>>>>>>>>>>>>
and also
>>>>>>>>>>>>>>
>content={ "xxxx" : { "result" : {"Channel" : "Temp1", "Unit" : "Temperature", "Float" : "1" , "Decimalmode": "2" , "Value" : "22
138195 : Command: content={
138196 : Command unknown
Unknown command!
>>>>>>>>>>>>>>
The string seems to be cut to a 128 characters string.
Could it be a reason for the crash in the following and last releases?

TD-er, 2018-08-18:
I am currently looking at all controllers to add a small queue option and using the scheduler to process the sending of their data.
I will then also look into the max. length of the data used to send.

It is very well possible the JSON parser is not that stable when parsing incomplete strings.

popaserge, 2018-08-23:
Hello

I tested the new release ESP_Easy_mega-20180822_dev_ESP8266_4096.bin
Same crash occurred with same message as previously.

Panic /home/travis/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/core_esp8266_main.cpp:103 __yield

For the only version not crashing that I know (ESP_Easy_mega-20180522-26-ga5f44a1_dev_ESP8266_4096.bin) I simplified all the syslog settings to "info" and set up the Message Interval to 30000 ms. Not so much info is coming to the serial port 
I could observe that this version also crashes with following message which is different.

Panic /home/gijs/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/core_esp8266_main.cpp:130 loop_task

This occured after following message (24 mn?)
2852605 : WD   : Uptime 48 ConnectFailures 0 FreeMem 18056

and then again after  
5732762 : WD   : Uptime 96 ConnectFailures 0 FreeMem 18056

5312476 : WD   : Uptime 89 ConnectFailures 0 FreeMem 18056

3212455 : WD   : Uptime 54 ConnectFailures 0 FreeMem 18056

992443 : WD   : Uptime 17 ConnectFailures 0 FreeMem 18056

2372442 : WD   : Uptime 40 ConnectFailures 0 FreeMem 18056

A new track to understand the crash?

TD-er, 2018-08-23:
"Message Interval to 30000 ms" 
That's one receipt for crashes on its own.
You better not set that value above 1 second.
The best setting for that value is 100 msec and I am working very hard to get that value removed at all, since it doesn't do what you think it should do.
See https://github.com/letscontrolit/ESPEasy/pull/1669

tonhuisman, 2023-09-03:
This seems to be solved, so can be closed.

closed 2023-09-04 by TD-er