letscontrolit/ESPEasy #1537: 48506 : LoadFromFile: config.dat index: 32768 datasize: 1024

dzaczek, 2018-06-28:
### Summarize of the problem/feature request

Error while reading/writing config.dat
               

### Expected behavior

I should add new controller and save configuration  .

### Actual behavior

I cant add new controller . Wemos D1 mini reboot after submit configuration  . And when i open again controller i se many different char in  configuration like this: "Ҁ�??I@XJ�?�N�?���?�@"

### Steps to reproduce
1. When i try add controller  save finish  reboot 
I tried on 2 wemos d1 mini  effect is the same 

### System configuration

Hardware:
Wemos d1 mini 

ESP Easy version: 
mega-20180625

ESP Easy settings/screenshots: 
Build
20102  - Mega
Libraries
ESP82xx Core 2_4_1, NONOS SDK 2.2.1(cfd48f3), LWIP: 2.0.3
GIT version
mega-20180625
Plugins
73 [Normal] [Testing] [Development]
Build time
Jun 25 2018 02:18:45
Binary filename
ESP_Easy_mega-20180625_dev_ESP8266_4096.bin

### Rules or log data

```
48526 : FS   : Error while reading/writing config.dat in 1066

```

TD-er, 2018-06-28:
The config file is a binary format, so seeing garbage is not an error.
Can you check the vendor ID of the flash chip?

And is it possible to gather a log from the moment you try to save, which leads to a crash?

dzaczek, 2018-06-28:
`
Detecting chip type... ESP8266
Chip is ESP8266EX
Chip ID: 0x004ef29e`
Restart when i change type from ip to hostname in Generic HTTP Advanced [TESTING] without submit data 
https://asciinema.org/a/FLWGJFm7tQhU0SVVxVJqsGAzS

`359425 : SaveToFile: config.dat index: 32768 datasize: 1024                                                                                                                                                                                                             Panic /home/travis/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/core_esp8266_main.cpp:99 __yield                                                                                                                                                                                                          ctx: sys                                                                       sp: 3fff3050 end: 3fffffb0 offset: 01b0                                                                                                                                                                                                                               >>>stack>>>                                                                                                                                    3fff3200:  00000100 3fff3260 3fffc718 0032bc04                                                                           3fff3210:  00000004 401051ee 40105734 4025c348                                                                                      3fff3220:  3fff3260 3fff3260 00000000 4025c378                                                                                      3fff3230:  3fffc71                           `
`
Second wemos 
similar situation 

Detecting chip type... ESP8266
Chip is ESP8266EX
Stub running...
Chip ID: 0x0073008a

`Exception (2):                                                                                                                      epc1=0x3ffeb92c epc2=0x00000000 epc3=0x00000000 excvaddr=0x3ffeb92c depc=0x00000000                                                                                                                                                                                     ctx: cont                                                                                                                                     sp: 3ffffddc end: 400043e6 offset: 01a0                                                                                                                                                                                                                       >>>stack>>>                    `

`93606 : SaveToFile: config.dat index: 32768 datasize: 1024                                                                          93628 : FS   : Error while reading/writing config.dat in 1005                                                                       93630 : SaveToFile: config.dat index: 28672 datasize: 724                                                                           93682 : FILE : Saved config.dat              `

TD-er, 2018-06-28:
I guess the problem is probably more related to the "Generic HTTP Advanced [TESTING]" and not saving the settings.
The error indicates a `yield()` problem, which is often an indication of calling either `yield()` or `delay()` from a call-back function.

dzaczek, 2018-06-29:
Maybe this is 2 issue i checked soft  mega-20180621and had problem with  garbage in fields HTTP URI: HTTP HEADERS: HTTP BODY:

480640 : WD   : Uptime 8 ConnectFailures 0 FreeMem 17912
510640 : WD   : Uptime 9 ConnectFailures 0 FreeMem 17912
540640 : WD   : Uptime 9 ConnectFailures 0 FreeMem 17912
570640 : WD   : Uptime 10 ConnectFailures 0 FreeMem 17912
600640 : WD   : Uptime 10 ConnectFailures 0 FreeMem 17912
630640 : WD   : Uptime 11 ConnectFailures 0 FreeMem 17912
639712 : SaveToFile: config.dat index: 32768 datasize: 1024
639731 : FS   : Error while reading/writing config.dat in 1005
639782 : SaveToFile: config.dat index: 28672 datasize: 724
639837 : FILE : Saved config.dat
639843 : SaveToFile: config.dat index: 0 datasize: 1220
639899 : FILE : Saved config.dat
639917 : LoadFromFile: config.dat index: 28672 datasize: 724
639950 : LoadFromFile: config.dat index: 32768 datasize: 1024
639969 : FS   : Error while reading/writing config.dat in 1066
640014 : Ram usage: Webserver only: 0 including Core: 0 flashStringCalls: 23 flashStringData: 7128

tonhuisman, 2022-04-18:
This seems to have been resolved, issue can most likely be closed.

closed 2022-04-18 by TD-er