letscontrolit/ESPEasy #1539: Please add a parameter to invert the values on thePCA9685 board

micropet, 2018-06-28:
With pcapwm, 0.0, the LED driver switches the LED to 100% brightness.
pcapwm, 0.4095 turns off the LEDs.
This is a little unattractive for me.

Can you still add a parameter to invert the values?

micropet, 2018-06-29:
A fade-in fade-out function would also be quite fine.

workgroupengineering, 2018-06-29:
I think it is better to add a setting in the configuration of the device that indicates if it is inverted or not.

micropet, 2018-06-29:
Good idea.

micropet, 2018-06-29:
Can you also make an adjustable fade-in and fad-out time?
Best as a commando.

M0ebiu5, 2018-06-29:
Fade-out time = 10 sec. But the led is off in 6 seconds - according to your first description...
What is your use case? "fading a led anyhow"?

micropet, 2018-06-29:
The best is an adjustable time. Maybe from 0.1 sec to 5 minutes

For normal operation in the apartment  you will choose a time of under one second.

A sunrise or sunset needs more time.

micropet, 2018-06-29:
@M0ebiu5,

i see no Fade-out time = 10 sec.
The LED is immediately ON and immediately OFF. No dimming.

ShardanX, 2018-07-01:
The "inverted" is a problem of settings for the chip.
I've described that in #651 in December '17 including a somewhat easy way to solve it.
The output is set to directly drive LED's - if you use an IGBT-driver for a "fat" MOSFET for example the output of the chip gives an inverted result (0= full bright, 4095 = off)
I'm not a coder but it should be not too difficult to make a checkbox in the GUI and set the register code
depending on that checkbox set/not set as described in #651

A well done fading is somewhat more complex. It is a count up / count down from a given pwm value to another one in a given time. 

For example: There is a LED driver set to PWM= 1000 and you want to dim up to full bright (PWM=4095) within 30 seconds.
1st. step: The start value must be read from the PCA.
This is necessary to avoid "flashing" - if you don't use the actually set value the start of dimming may jump from the actual value to a manually given start value destroying the fading effect.

2nd. step: Calculate number of necessary steps (4095 - 1000 = 3095 steps)

3rd. step: Calculate the time step between single steps (30s / 3095steps ~ 10 ms)

4th. step: Count up from 1000 to 4095 in 10 ms steps...

The way for dimming down is same, just count down instead up.

I doubt if it is that easy to code inside ESPeasy.

M0ebiu5, 2018-07-02:
> 3rd. step: Calculate the time step between single steps (30s / 3095steps ~ 10 ms)

In case of espeasy, where you cannot dispose over interrupts, you would calculate the step size until next callback (PLUGIN_FIFTY_PER_SECOND is the highest resolution currently possible i guess).

Anyway, i think a good solution would have to consider some more aspects like:
- hardware independence (so that it works with internal gpios and any other plugin also)
- an abstract definition of the brightness (eg 0-100% in 0.5% steps)
- a lookup table or calculation routine to map power to brightness (so that 50% looks as half as bright as 100%)
- a configurable pwm range with holes at start and end (certain boundary pwm levels cause flicker)
- etc

Maybe - before hacking some naive, quick code - it's worth to think about how to do it right and provide a more elegant and universal implementation?

ShardanX, 2018-07-03:
Agree. My posting just was meant as some clarification in general.
Anyways, I see the "Inverted/non-inverted" as more important as this influences hardware massively.
I do dimming from my home controller, but the hardcoded setting makes hardwaredesign more complex then necessary.

M0ebiu5, 2018-07-03:
@ShardanX i see your point and my post was mostly addressed to op (who likes to waste everybody's time by not mentioning version numbers or that he runs a self compiled build and now by making vague fr's without answering any question). Sorry, but that just pushes my buttons...

What kind of home controller and hardware do you use? I have some rcd-24, ldd's (gpio or pca controlled) and led strips with ws2812/11 in use here.

ShardanX, 2018-07-05:
Atm I use FHEM as a home control and have some cheap china controllers with standard RGBWW strips.
Controller: https://de.aliexpress.com/item/Led-RGB-Wifi-Controller-DC12-24V-Magic-UFO-Wifi-Led-Controller-With-21Key-RF-Controller-For/32818725093.html?spm=a2g0x.search0104.3.26.13063adanTHCFW&ws_ab_test=searchweb0_0,searchweb201602_3_10320_10152_10151_10065_10321_10344_10068_10342_10547_10343_10322_10340_10548_10341_10696_10084_10083_10618_10304_10307_10820_10821_10302_10843_10059_100031_10319_10103_524_10624_10623_10622_10621_10620,searchweb201603_1,ppcSwitch_5&algo_expid=48ede431-9d69-4112-8388-2e03731de287-4&algo_pvid=48ede431-9d69-4112-8388-2e03731de287&priceBeautifyAB=0

The controller can be integrated into FHEM directly so it is very easy to use. Nevertheless it has some disadvantages, tends to flicker and flash and such so I'm searching for another solution, ESPeassy preferred. Hardware isn't much of a problem but a switchable inverting would make things less complex.

closed 2018-10-22 by micropet