letscontrolit/ESPEasy #1540: MQTT Import: Allow string messages

codegrau, 2018-06-29:
Allow to receive string messages for connected displays (OLED,LCD,...) with MQTT Import Plugin

TD-er, 2018-06-29:
At this moment, the MQTTimport is just a plugin, which means it can only output float values, like any other plugin.
This means there should be some kind of extra layer in ESPeasy to collect data and feed it to some rule engine, or create events.

codegrau, 2018-06-29:
I‘d really appreciate that. Would make it a lot easier to build user interfaces without the need of additional stuff besides ESPeasy

Grovkillen, 2018-06-30:
Yes we need to not mix up values (float) with strings. All device values should always be float. Strings are something new and we need to look really close to where to put it. I personally would love to have strings to mess around work on but it should not be through device values.

uzi18, 2018-06-30:
I see it different way, to give mqtt import command execution support

Grovkillen, 2018-06-30:
It should be part of a different layer (as stated be @TD-er) I think. Given that you are working with it in a completely different way than with float values we should not mix them. But lets leave the for later :)

TD-er, 2018-07-01:
@uzi18 I do not entirely understand your post.
Maybe you could clarify it a bit?

uzi18, 2018-07-01:
for example add checkbox to not set value but execute command with argument from imported variable

TD-er, 2018-07-01:
@uzi18 
My idea of the future of MQTT import is to remove it as a plugin.
After all, the controllers already have the role to keep the MQTT connection active to a broker, so why not use that connection for incoming data.
This also makes it possible to allow (much) more than the current number of imports. (subscribes to topics)
And maybe there are other controller-protocols that could benefit from similar functionality.

This means the plugins supporting it should have an extra option to select the "controller in" (all existing controllers are "out").
I am not sure yet where the subscribe string should be defined. I guess it is somewhat controller specific, so there could be a new breed of controllers which store a number of subscribe strings.

When such a topic receives a message, all subscribed to that topic should receive an event (rules included) which will trigger appropriate actions.
For these there should be no limitation on number of parameters and type (string/float/bool), since the plugin handling it will have to parse it.
I think only a limited number of plugins may benefit from this and maybe we could add some extra setting to allow the commands to be sent via one or more topics. (on one or more controllers)

That's about what I had in mind for the future of MQTT import.

coolduke666, 2018-09-24:
sounds great... hope it will be implemented soon :)

Obicom, 2019-02-15:
Is there any use? I expected that it still work after I saw this blog : https://community.home-assistant.io/t/lcd-display-with-mqtt-and-home-assistant-notifies/33478

But still have the problem that I am not able to import text payloads ... any advice would be highly appreciated ... Thanks in advance

Grovkillen, 2019-02-15:
No strings are allowed for values. It's going on stuff behind the scenes but no strings are yet supported.

But you can use values that are setting the display text. It's not dynamic though. Depends on what you want to accomplish?

Grovkillen, 2019-02-15:
But you are able to use these types of commands for strings:

http://192.168.1.91/control?cmd=lcd,4,6,DEACTIVATED!

Obicom, 2019-02-15:
@Grovkillen that was also my idea to work with rules to translate the values to text but do you have an example? I am total new with ESP Easy  I want to send then a number als value from Home Assistant and would like to see text on an OLED display like "Window Open".

Obicom, 2019-02-15:
Is there a reason why ESP Easy limit mqtt topic only to numbers? I guess this is not a mqtt standard limitation ...

Grovkillen, 2019-02-15:
Topics are not numbers only. (Task) values are only numbers.

You can use the cmd topic for your unit and send something like this as payload (message):
lcd,4,6,DEACTIVATED

Obicom, 2019-02-15:
@Grovkillen I have now an OLED display and tried to send commands via http but nothing happens. Is this related to the admin password? Can I transfer the password with the URL? (http://192.168.1.95/control?cmd=OLEDCMD,off)

Grovkillen, 2019-02-15:
Yes, if you have admin password enabled you will not get those HTTP requests through. Use MQTT instead.

Obicom, 2019-02-15:
Then I am back to my first concern .. only numbers are allowed ... no text ... grrrr

Grovkillen, 2019-02-15:
No, you can use the same commands thorough MQTT (as I stated before).

Publish to the "cmd" topic of your device (and please read the wiki...

Obicom, 2019-02-15:
Ok I found and it works ... but only with the normal OLED ... not with framed OLED. Normal OLED has very small font ... framed OLED would be perfect ... :(

Grovkillen, 2019-02-15:
I answered you on the forum as well. Please use that thread for further discussions.

MC-1984, 2020-08-26:
Thank you for your very detailed explanation. 

When I ping the ESP some requests end up with a timeout

`PING 192.168.50.235 (192.168.50.235): 56 data bytes
64 bytes from 192.168.50.235: icmp_seq=0 ttl=254 time=131.751 ms
64 bytes from 192.168.50.235: icmp_seq=1 ttl=254 time=48.214 ms
64 bytes from 192.168.50.235: icmp_seq=2 ttl=254 time=274.413 ms
64 bytes from 192.168.50.235: icmp_seq=3 ttl=254 time=190.023 ms
64 bytes from 192.168.50.235: icmp_seq=4 ttl=254 time=110.022 ms
64 bytes from 192.168.50.235: icmp_seq=5 ttl=254 time=30.746 ms
64 bytes from 192.168.50.235: icmp_seq=6 ttl=254 time=258.824 ms
64 bytes from 192.168.50.235: icmp_seq=7 ttl=254 time=174.656 ms
64 bytes from 192.168.50.235: icmp_seq=8 ttl=254 time=95.548 ms
64 bytes from 192.168.50.235: icmp_seq=9 ttl=254 time=328.125 ms
64 bytes from 192.168.50.235: icmp_seq=10 ttl=254 time=240.253 ms
Request timeout for icmp_seq 11
64 bytes from 192.168.50.235: icmp_seq=12 ttl=254 time=79.663 ms
64 bytes from 192.168.50.235: icmp_seq=13 ttl=254 time=308.337 ms
64 bytes from 192.168.50.235: icmp_seq=14 ttl=254 time=228.057 ms
Request timeout for icmp_seq 15
64 bytes from 192.168.50.235: icmp_seq=16 ttl=254 time=70.198 ms
Request timeout for icmp_seq 17
64 bytes from 192.168.50.235: icmp_seq=18 ttl=254 time=221.075 ms
64 bytes from 192.168.50.235: icmp_seq=19 ttl=254 time=146.903 ms
64 bytes from 192.168.50.235: icmp_seq=20 ttl=254 time=63.740 ms
64 bytes from 192.168.50.235: icmp_seq=21 ttl=254 time=290.978 ms`

Can you give me another hint how to fix this? I already tried all wifi settings on the wifi router. But it had no effect.

TD-er, 2020-08-26:
This last question is becoming quite off topic and more a forum question.
Like I said, have a look at the timing stats page to see if there are calls made to some function that may take quite a long time.
Those are blocking calls which may also block WiFi related code like answering to pings etc.

To continue those connectivity related questions, please open a topic about it on the forum as Github issues are about bugs/changes or new features to implement and the forum is about operating the software

MC-1984, 2020-08-26:
I will do that. Thank you again!

MarkusH1975, 2023-02-27:
@TD-er 
Is there a plan to support string variables?
Receive string from MQTT and show the string on e.g. an OLED Display.
(Not mapping strings to numeric values.)

tonhuisman, 2023-02-27:
You can handle that from the MQTT Import plugin like this:
Enable `Generate events for accepted topics` checkbox
```
// For regular MQTT topics:
On MQTTImport#Topic=* Do
  OLedFramedCMD,1,12,"%eventvalue1%"
Endon
// Or for JSON messages:
On Topic#variable=* Do
  OLedFramedCMD,1,12,"%eventvalue1%"
Endon
```

NB: The `*` in the `On ... Do` line makes it accept non-numeric values as the first argument, you will need a somewhat recent release of ESPEasy for that to work.

MarkusH1975, 2023-03-16:
@tonhuisman thanks for the hint, this helped my to find a solution. My solution looks a bit different. 
(this may help others)

**Generic - MQTT Import device** 

- Name "mqttImport"
- Generate events for accepted topics 'true'
- Prefix for all topics "%sysname%/oled1/"
- MQTT Topic "lineA"
- MQTT Topic "lineB"
- MQTT Topic "lineC"
- MQTT Topic "lineD"

**Display - OLED SSD1306/SH1106 Framed device**

- Name "oled1"
- 12 Lines used
- 1-8 used with local content
- 9-12 used for mqtt content, lineA-lineD

**Rules for filling oled with mqtt content**

```javascript
// Example:
// mosquitto_pub -h 192.168.x.y -t "MySysname/oled1/lineA" -m "MyString for LineA"

// NameOfMqttImportDevice#topicPrefix/TopicName

On mqttImport#%sysname%/oled1/lineA=* Do
  // OLED device name is "oled1"
  // set line number 9 with string content of lineA mqtt topic
  [oled1].oledframedcmd,9,"%eventvalue1%"
Endon
  
On mqttImport#%sysname%/oled1/lineB=* Do
  [oled1].oledframedcmd,10,"%eventvalue1%"
Endon

On mqttImport#%sysname%/oled1/lineC=* Do
  [oled1].oledframedcmd,11,"%eventvalue1%"
Endon

On mqttImport#%sysname%/oled1/lineD=* Do
  [oled1].oledframedcmd,12,"%eventvalue1%"
Endon

``` 

**Events in log**

Event exampe visible in log:
EVENT: mqttImport#MySysname/oled1/lineA='My line A string'
ACT : [oled1].oledframedcmd,9,'My line A string'

tonhuisman, 2023-09-01:
This issue seems to be solved, can be closed.

closed 2023-09-01 by TD-er