letscontrolit/ESPEasy #1542: Domoticz IDX more than 4 characters please

basjes1977, 2018-07-01:
Using the test build 20180625 and earlier it is not possible to set a IDX for domoticz device larger than 4 numbers. It throw an error staking idx cannot be larger than 9999. Unfortunately my IDX-es are higher and are 5 digits long.

Would it be possible to enlarge the accepted IDX size to 5 characters?
My domoticz database is several years old and as far is i know it does not reuse idx-es of removed devices. So it would be nice to have for me.
Thanks.

Hardware: Adafruit Feather Huzzah with ESP8266 https://www.adafruit.com/product/2821

ESP Easy version: mega-20180625

Grovkillen, 2018-07-01:
As a workaround you can use rules. It will be fixed though, time frame not set yet.

basjes1977, 2018-07-01:
Thanks for pointing me to the workaround👍🏻

closed 2018-07-30 by TD-er