letscontrolit/ESPEasy #1544: webGUI new menu entry

enesbcs, 2018-07-01:
Will it be possible to create a new main menu entry on ESPEasy webGUI?
I'm thinking about a plugin that implements a new page, but i am unable to modify the main menu, because it's entries declared locally in void getWebPageTemplateVar(const String& varName ) as "static const __FlashStringHelper* gpMenu[8][2]".
If gpMenu can be declared somewhere globally, maybe it can be accessed by a plugin.

Grovkillen, 2018-07-01:
I'm doing a complete overhaul of the webGUI together with a lot of improvements in customisation.

First off, it will be themable
It will be possible to add custom pages (own HTML CSS JS)
It will contain a dashboard for task values, with graphs etc.
And more

It will be completely javascript built and thus take much of the muscle work OFF the ESP and leave that to the browser. We can make a lot more eye candy that way. So to answer your question; yes you will be able to customize colors and looks in the generic webGUI and make what ever you like with single file HTML (inline CSS and JS).

sparkescreative, 2018-07-02:
Looking forward to this.

tonhuisman, 2023-09-02:
No relevant information here AFAICS, so can be closed.

closed 2023-09-02 by TD-er