letscontrolit/ESPEasy #1545: Pulse counter do not send Total and Time

swiezaks, 2018-07-02:
### Summarize of the problem/feature request

Using pulse counter on every espeasy (tried few version) esp sending only first value count. do not sending total and time. trying mqtt and generic http advanced.

### Expected behavior

esp should sending 3 values: count, total, time

### Actual behavior

send only one value- count.

### Steps to reproduce

1. 
2. 
3. 

### System configuration

Hardware: 
wemos d1 mini, 
and logs
https://obrazki.elektroda.pl/2352461800_1530532580.png

ESP Easy version: 
mega-20180624

ESP Easy settings/screenshots: 
https://obrazki.elektroda.pl/7547759200_1530532580.png
https://obrazki.elektroda.pl/2352461800_1530532580.png

Nerd522, 2018-07-02:
Change your counter type to /Delta/Total/Time and you get all 3 values.

Grovkillen, 2018-07-02:
Ahh, that's right. But the plugin shouldn't display the other values then so we'll leave it as a reminder to fix that plugin.

swiezaks, 2018-07-02:
thank you very much!

tonhuisman, 2023-09-21:
AFAICS, this issue has been resolved, the reminder is no longer needed, and the GPIO timing/handling (including this plugin) has been greatly improved over the last 6 to 8 months.

closed 2023-09-21 by TD-er