letscontrolit/ESPEasy #1547: not all events or status changes in log

TungstenE2, 2018-07-03:
### Summarize of the problem/feature request

Not all events or status changes appear in log. No matter what log level I choose, only random events or status changes triggered by rules appear in log.

### Expected behavior

All events and status changes should appear in log.

### Actual behavior

see above

### System configuration

Hardware: Wemos D1 mini

ESP Easy version:   Build 20180621

Grovkillen, 2018-07-03:
You talking about the web log?

TungstenE2, 2018-07-03:
talking about Tools -> 'Open log output'

Grovkillen, 2018-07-03:
So yes, that's the web log. We are experiencing some log lines dropout. I will investigate this further once I get back from my holidays.

TungstenE2, 2018-07-03:
thx. I would say less than 10% are displayed. Let me know if you need more details.

Grovkillen, 2018-07-03:
I would really appreciate if you could do a parallel check with serial log and see if there's a pattern of the missing entries.

TD-er, 2018-07-03:
If you think less than 10% is displayed, then please have a look at the log settings.
For each output (serial, syslog, web) you can set the max. log level (info, error, debug, debug more)
Please make sure both views you compare are set to the same level.

The bug @Grovkillen mentioned was about some requests lost between browser and the ESP node.
So that's just a few lines of logs. You can see some gap in msec in front of each line.
At most every second the logs are fetched by the browser. A lost update will lead to a gap of at most 1 second (1000 msec) between log lines.  (given there were logs in that second)

TungstenE2, 2018-07-03:
I set Syslog Level, Serial log Level and Web log Level to 'Info'.
Assuming that device status changes of switches and events are in the log.

How to check the serial log if IP/log is only the weblog?

TD-er, 2018-07-03:
Not sure I understand your last remark.

The serial log can only be checked via a serial console (and serial connection via the USB adapter used to flash the initial firmware) from tools like Arduino IDE, Atom or VScode + PlatformIO.

TungstenE2, 2018-07-03:
thx, got it working with Visual SYSLOG server for windows.

I can confirm I receive the status changes and events with serial log but only partialy in web log.

s0170071, 2018-07-03:
see https://github.com/letscontrolit/ESPEasy/issues/1523

TD-er, 2018-07-03:
Since it appears to be a duplicate of #1523, I will close this one.

closed 2018-07-03 by TD-er