letscontrolit/ESPEasy #1549: [info page] Add a page to the web-interface showing all system variables

TD-er, 2018-07-04:
Currently we have a page on the [wiki](https://www.letscontrolit.com/wiki/index.php/ESPEasy_System_Variables) describing the system variables.
However, this list can get outdated and it never shows the current values of your own system.

Also that list can become outdated, or show variables not yet implemented in the running firmware version.

Add a page to the web interface showing a list of all the system variables and their current values.

Grovkillen, 2018-07-04:
This is a nice one. Make a JSON fetch link and I'll make sure it's in the new GUI

uzi18, 2018-07-04:
Nice idea, it will resolve lot's of questions.

Grovkillen, 2018-07-04:
+ that I can use it for future rules IDE

TungstenE2, 2018-08-30:
any news on this? would be a great feature

Grovkillen, 2018-08-30:
We're currently focusing on the new queue for the controllers. Once that is released stuff like this will be in our loop. I'm gonna pick up speed regarding the GUI overhaul I'm doing and for that one we want this to be from day one :)

TD-er, 2018-09-02:
Implemented and merged

closed 2018-09-02 by TD-er