letscontrolit/ESPEasy #1552: Participants in P2P setup don't see all others

ericdes, 2018-07-05:
### Summarize of the problem/feature request

My setup is with 3 Sonoffs Basic participating in the same P2P setup (UDP port is the same). A device might list all the nodes while another might list only itself or itself and another one, The problem occurs erratically, so far I haven't been able to have all devices see all others.

### Expected behavior

YOUR TEXT GOES HERE

### Actual behavior

YOUR TEXT GOES HERE

### Steps to reproduce

1. 
2. 
3. 

### System configuration

Hardware:

ESP Easy version: mega-20180625

ESP Easy settings/screenshots: 

### Rules or log data

```

```

TD-er, 2018-07-06:
Things to check for each ESP node:

- Unique unit number
- Use same UDP port
- Perform reboot of ESP node after setting UDP port
- All on the same network
- IP filtering (in ESPeasy settings) not blocking some nodes

Can you confirm all these are OK?

ericdes, 2018-07-06:
Yes, all these are OK.

TD-er, 2018-07-06:
Are they connected to the same accesspoint?
Using a guest network, or normal setup?

In some guest network setups the communication between wifi clients is prohibited.

Could you list their IP addresses and subnet?

ericdes, 2018-07-06:
They’re connected to the same access point, using the normal setup.

 

I see their IP addresses in the DHCP client list in the router.

TD-er, 2018-07-06:
And are those IP addresses the same as reported in ESPeasy in the sysinfo page?
And all on the same subnet?

Wiki591, 2018-07-14:
Hi,
I tried to use P2P, too. 
Four Nodes are running on mega2018-05-22, my central one is running on mega2018-05-24. The central one I want to put an oled on to have a small monitoring display.
First setup worked, the devices of the four nodes showed up in my central node (and in the other ones, too). Only the order of the devices was mixed up. So I deleted the devices on my central node, switched off the ESPEasy P2P Networking controller, reconfiguered the productive nodes to get a correct ordered list of the devices in the other four nodes and switched P2P controllers on again.
Now no device from no node shows up in my central Node. Additionally, switching P2P controlller on leads to instable behaviour of the nodes. They are rebooting randomly between 6-24 hours. I have switched off the P2P again, everything is working stable now.

Configuration:
All in one subnet, on one AP
All the same UDP-Port
The four productive nodes on DHCP, the central node with fixed IP
The nodes all show up in the nodelist of the main tab of each node with their correct IP address

Your questions:

Things to check for each ESP node:

    Unique unit number
    Use same UDP port
    Perform reboot of ESP node after setting UDP port
    All on the same network
    IP filtering (in ESPeasy settings) not blocking some nodes

Can you confirm all these are OK?

Yes, all these are OK.

Grovkillen, 2019-10-10:
Reopen if still an issue.

closed 2019-10-10 by Grovkillen