letscontrolit/ESPEasy #1555: TSL2591 values are not transferred to controller

barthm1, 2018-07-08:
Sensor values are shown in devices tab - The values are not transferred to Controller - tested thingspeak and generic http.

s0170071, 2018-07-08:
Send to controller check marks in devices tab are set?
Device Delay set?
Controller is enabled?
In advanced settings set delay to 100. At to of page.
Obey thingspeak 15 sec update limit?

barthm1, 2018-07-09:
Send to controller check marks in devices tab are set? - yes
Device Delay set? - yes
Controller is enabled? - yes
In advanced settings set message interval to 100. At top of page.  - Integration time? - is 100 ms
Obey thingspeak 15 sec update limit? - yes

s0170071, 2018-07-09:
Can you set the log level to debug and post the serial log?

barthm1, 2018-07-09:
INIT : Booting version: mega-20180708 (ESP82xx Core 2_4_1, NONOS SDK 2.2.1(cfd48f3), LWIP: 2.0.3)
89 : INIT : Warm boot #1 - Restart Reason: External System
93 : FS   : Mounting...
120 : FS   : Mount successful, used 75802 bytes of 957314
463 : CRC  : program checksum       ...OK
497 : CRC  : SecuritySettings CRC   ...OK 
604 : INIT : Free RAM:21056
605 : INIT : I2C
605 : INIT : SPI not enabled
618 : TSL2591: Address: 0x29: Integration Time: 100 ms Gain: 1x (Low)
629 : INFO : Plugins: 71 [Normal] [Testing] (ESP82xx Core 2_4_1, NONOS SDK 2.2.1(cfd48f3), LWIP: 2.0.3)
630 : WIFI : Set WiFi to STA
662 : WIFI : Connecting RWLAN2 attempt #0
783 : LoopStats: shortestLoop: 10000000 longestLoop: 0 avgLoopDuration: 0.00 systemTimerDuration: 0.00 systemTimerCalls: 1 loopCounterMax: 1 loopCounterLast: 1 countFindPluginId: 1
791 : WD   : Uptime 0 ConnectFailures 0 FreeMem 20008
1266 : TSL2591: Lux: 2.04 Full: 2.00 Visible: 1.00 IR: 1.00
4510 : WIFI : Connected! AP: RWLAN2 (XX:XX:XX:XX:XX:XX) Ch: 5 Duration: 3847 ms
6432 : WIFI : DHCP IP: 192.168.1.88 (deviot88-88) GW: 192.168.1.202 SN: 255.255.255.0   duration: 1922 ms
6446 : Webserver: start
6828 : Current Time Zone:  DST time start: 2018-03-25 02:00:00 offset: 120 minSTD time start: 2018-10-28 03:00:00 offset: 60 min
11143 : TSL2591: Lux: 2.04 Full: 2.00 Visible: 1.00 IR: 1.00
21144 : TSL2591: Lux: 2.04 Full: 2.00 Visible: 1.00 IR: 1.00
31143 : TSL2591: Lux: 2.04 Full: 2.00 Visible: 1.00 IR: 1.00
31156 : LoopStats: shortestLoop: 46 longestLoop: 885179 avgLoopDuration: 84.19 systemTimerDuration: 26.41 systemTimerCalls: 32 loopCounterMax: 652173 loopCounterLast: 352132 countFindPluginId: 0
31175 : WD   : Uptime 1 ConnectFailures 0 FreeMem 13600
41143 : TSL2591: Lux: 2.04 Full: 2.00 Visible: 1.00 IR: 1.00
51143 : TSL2591: Lux: 2.04 Full: 2.00 Visible: 1.00 IR: 1.00
61144 : TSL2591: Lux: 2.04 Full: 2.00 Visible: 1.00 IR: 1.00
61176 : LoopStats: shortestLoop: 46 longestLoop: 885179 avgLoopDuration: 93.19 systemTimerDuration: 25.97 systemTimerCalls: 31 loopCounterMax: 652173 loopCounterLast: 315725 countFindPluginId: 0
61193 : WD   : Uptime 1 ConnectFailures 0 FreeMem 13424
71143 : TSL2591: Lux: 2.04 Full: 2.00 Visible: 1.00 IR: 1.00
81143 : TSL2591: Lux: 2.04 Full: 2.00 Visible: 1.00 IR: 1.00
91143 : TSL2591: Lux: 2.04 Full: 2.00 Visible: 1.00 IR: 1.00
91194 : LoopStats: shortestLoop: 46 longestLoop: 885179 avgLoopDuration: 89.01 systemTimerDuration: 25.71 systemTimerCalls: 31 loopCounterMax: 652173 loopCounterLast: 330210 countFindPluginId: 0
91211 : WD   : Uptime 2 ConnectFailures 0 FreeMem 13424
96510 : FILE : Saved config.dat
96577 : FILE : Saved config.dat
96585 : TSL2591: Address: 0x29: Integration Time: 100 ms Gain: 1x (Low)
101143 : TSL2591: Lux: 2.04 Full: 2.00 Visible: 1.00 IR: 1.00
121212 : LoopStats: shortestLoop: 46 longestLoop: 885179 avgLoopDuration: 88.78 systemTimerDuration: 25.55 systemTimerCalls: 31 loopCounterMax: 652173 loopCounterLast: 331006 countFindPluginId: 0
121230 : WD   : Uptime 2 ConnectFailures 0 FreeMem 11664
131143 : TSL2591: Lux: 0.00 Full: 1.00 Visible: 0.00 IR: 1.00

s0170071, 2018-07-10:
Looks ok but I can't see any connection attempts. Can you post the controller settings please?

TD-er, 2018-07-10:
I looked into the source code of that plugin and I don't see a `success = true;` line in the plugin read case.
Please add that statement somewhere in these lines:
https://github.com/letscontrolit/ESPEasy/blob/b9f3f683358fc1cc17e897a4588a3603dd6828ed/src/_P074_TSL2591.ino#L246-L255

Let me know if that is the solution and then I will change it asap.

barthm1, 2018-07-11:
Yes - works now

closed 2018-07-11 by barthm1