letscontrolit/ESPEasy #1556: SendToHTTP not working

yuckinus, 2018-07-08:
need some help here please.

ESPEasy_mega-20180708
Running on a Sonoff TH16 with ESP8266/1024

Trying to load the data to wunderground so I see I can either user Generic HTTP controller or a rule.

With Generic HTTP controller I seem to be able to do basic uploading as long as the Controller Publish text is not too long, like:
`/weatherstation/updateweatherstation.php?ID=NAME&PASSWORD=PASSWORD&dateutc=now&%valname%=%value%&action=updateraw`

Sadly this means I can't use their RapidFire update method which needs something like:
`/weatherstation/updateweatherstation.php?ID=NAME&PASSWORD=PASSWORD&dateutc=now&%valname%=%value%&realtime=1&rtfreq=10&action=updateraw`

If I try to use a rule it always crashes and reboots, either using a short or a long Controller Publish, like:
```
On System#Boot do    //When the ESP boots, do
   timerSet,1,30      //Set Timer 1 for the next event in 10 seconds
endon
 
On Rules#Timer=1 do  //When Timer1 expires, do
   SendToHTTP weatherstation.wunderground.com,80,/weatherstation/updateweatherstation.php?ID=NAME&PASSWORD=PASSWORD&dateutc=now&tempf=[Dummy#tempf]&action=updateraw
   timerSet,1,10       //Resets the Timer 1 for another 10 seconds
endon

```
Am I using SendToHTTP wrong? Did anybody use with succes data uploading to wunderground?

TD-er, 2018-07-12:
Just as a test, could you add an extra "," at the end of the line? 
There is one bug for sure in the `Command_HTTP_SendToHTTP` function.
The found index can be negative and there is no check for that, but it is used and interpreted as an unsigned int. (thus resulting in index 2^32)

Another strange thing is the implementation of `getParamStartPos`
And that's why I suggest adding a bogus "," at the end of the line in the rules just to test.

TD-er, 2018-07-12:
The crash is initiated by the watchdog timer:
```
10894 : ACT  : SendToHTTP weatherstation.wunderground.com,80,/weatherstation/updateweatherstation.php?ID=NAME&PASSWORD=PASSWORD&dateutc=now&tempf=&action=updateraw
10916 : LoadFromFile: config.dat index: 4096 datasize: 472
10925 : Command: sendtohttp

 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
v614f7c32
~ld
�U88 :

INIT : Booting version:  (ESP82xx Core 2_4_1, NONOS SDK 2.2.1(cfd48f3), LWIP: 2.0.3)
```

yuckinus, 2018-07-13:
sorry but for now I'm out of my workshop so can't test anything

Budman1758, 2018-07-22:
I tested a bunch of releases and standard non-controller related SendtoHttp worked fine until release mega-20180720. There are no controllers active here.  I use the command to tell a RPI Cam Control (Raspberry Pi camera) install to start and stop recording. Worked fine up to mega-20180719. Stopped working at mega-20180720 and up.  Is this the same or should this be a separate issue?

This is the exact command used in a rule...
`SendToHTTP 192.168.1.12,80,/html/cmd_pipe.php?cmd=ca%201`

ESP_Easy_mega-20180720_normal_ESP8266_4096   flashed to a generic ESP-01 with a 4 meg flash chip grafted on. (Makes OTA easy)

TD-er, 2018-07-22:
That's probably related to the parseString changes.
@grovkillen also reported the set WiFi SSID command did not accept a space anymore.

Budman1758, 2018-07-22:
So is it a bug or does the command need to be modified?

TD-er, 2018-07-22:
Not sure yet.
I will have to look into it to what's causing it.

TD-er, 2018-07-22:
It is a bug and I found it and made a fix for it.
I will also add the option to use quotes for allowing a space or comma in commands (e.g. in an URL or for displaying text on a display).
So there will be a fix for the last mentioned issue.
Not yet for the original issue in this topic.

Budman1758, 2018-07-23:
SendtoHTTP command is now working with mega-20180723.  :+1:

Grovkillen, 2018-07-23:
Thanks for the feedback!

TD-er, 2018-07-23:
I am not sure the original issue is resolved.

Grovkillen, 2018-07-23:
Aha, sorry! Let's reopen it.

yuckinus, 2018-08-07:
finally got the chance to test again my scenario, this time with mega-20180804. the good news is that the esp no longer crashes but the command doesn't seem to get executed either

`53369 : ACT  : SendToHTTP weatherstation.wunderground.com,80,/weatherstation/updateweatherstation.php?ID=IPANTELI3&PASSWORD=mjjs1kg7&dateutc=now&tempf=85.10&action=updateraw
53387 : GetArgv Error; argv_size exceeded. argc=4 argv_size=80 argv=/weatherstation/updateweatherstation.php?ID=IPANTELI3&PASSWORD=mjjs1kg7&dateut
53390 : GetArgv Error; argv_size exceeded. argc=5 argv_size=80 argv=/weatherstation/updateweatherstation.php?ID=IPANTELI3&PASSWORD=mjjs1kg7&dateut
53402 : GetArgv Error; argv_size exceeded. argc=6 argv_size=80 argv=/weatherstation/updateweatherstation.php?ID=IPANTELI3&PASSWORD=mjjs1kg7&dateut
53415 : Command: sendtohttp
53417 : SendToHTTP: Host: weatherstation.wunderground.com port: 80
53423 : SendToHTTP: Path: /weatherstation/updateweatherstation.php?ID=IPANTELI3&PASSWORD=mjjs1kg7&dateutc=now&tempf=85.10&action=updateraw
53654 : GET /weatherstation/updateweatherstation.php?ID=IPANTELI3&PASSWORD=mjjs1kg7&dateutc=now&tempf=85.10&action=updateraw HTTP/1.1
Host: weatherstation.wunderground.com
Connection: close`

from what I see there is a limit of 80 chars for the sendtohttp command but that's to small to work with wunderground. Can this be increased to something more reasonable like 128 or more?

TD-er, 2018-08-07:
I am looking into this issue: https://github.com/letscontrolit/ESPEasy/issues/1556
It is somewhat related to this issue.

For [this issue](https://github.com/letscontrolit/ESPEasy/issues/1625) I changed how the commands are being scheduled.
For this I just made this [build](https://www.dropbox.com/s/nxats6wfgdasj6g/ESPEasy_mega-20180804-11-g7783cf4.zip?dl=0) Can you test that build to see if it did change something regarding this issue?

yuckinus, 2018-08-07:
hmm.. actually I take it back. even if the log shows what appears to be an error message "GetArgv Error; argv_size exceeded. argc=4 argv_size=80", in the end the SendToHTTP command is executed just fine with "GET bla_bla HTTP/1.1"

TD-er, 2018-10-02:
It looks like this one can also be closed.
See also #1830

closed 2018-10-02 by TD-er