letscontrolit/ESPEasy #1559: EasyESP Mega ds18b20 MQTT

sevelen, 2018-07-11:
Hi, I use a NodeMcu with ESP_Easy_mega-20180710_normal_ESP8266_4096.bin

I´ve a problem with DS18b20 send via MQTT in combination with Pulse Counter. Please see below my tests. It looks like there is a bug inside. 

1 Test) I test just two task. Pulse + DS18b20. Pulse will be send the DS18b20 not (checked via MQTT.fx scanning)

2 Test) As next i disable the pulse counter. With the same result no DS18b20 data. 

3 Test) Just define one DS18b20 without any Pulse counter and the data will be send (MQTT.fx show them) 

4 Test) Task 1 = DS18b20 Task 2 = Pulse Counter result = Both sending 

5 Test) Task 1 = DS18b20 Task 2 = Pulse Counter Task 3 = Pulse counter Task 4 = DS18b20 Result first 3 Task will be send but Task 4 not

6 Test) Task 1 and 2 = Pulse Counter Task 3 - 6 DS18B20 Task 7 = BMP280 Task 8 and 9 DS18B20 Result: Pulse counter and BMP280 send but all DS18B20 not.

TD-er, 2018-07-11:
Just to prevent you doing a lot of other tests.
Make sure not to set "timing critical" plugins to other task numbers than 1...3 (or maybe 4).
There is still some limit to let timing critical plugins only run from the first few positions.

And this looks a lot like another reported issue about >3 values to be sent to some controller.

sevelen, 2018-07-12:
Any workaround for that issue?

sevelen, 2018-07-12:
I´ve checked the new Release mega-20180712 same issue DS18b20 will not send via MQTT: 

Log message: 
`EVENT: Heizung_Sole_Ruecklauf#Temperature=26.50
257880: EVENT: Heizung_Sole_Ruecklauf#Temperature=26.50 Processing time:3 milliSeconds
258850: Invalid value detected for controller OpenHAB MQTT`

TD-er, 2018-07-12:
This is a possibility this may not really be a bug, but designed to do so.
These 1-wire sensors have some error states (often referred to as reporting "85 degree C") and ESPeasy blocks those invalid values.
So there is somehow a fault in the conversion. This can be an initialization issue, noisy communication or some other issue. 
N.B. Other sensors may also produce NaN values, so it is not specific to these 1-wire modules.

It could also be there is somehow, somewhere a bug introduced or some strange setting which makes  the controller "thinks" it has to send more than one value, which somehow is an invalid value and this may stop sending any value to a controller.
If someone else could reproduce these issues, that would be nice.

sevelen, 2018-07-12:
Yes i know the Error states. But the values Shows correct at the Website and as well if i define them as Task 1.

On the other hand the same sensor wire worked serval years by an rsp.

Hopefully anybody can veryfies the issue soon thx

TD-er, 2018-07-12:
I know it shows a credible value, but if for some reason the plugin now wants to send >1 value and only one is present and initialized, then the hand-over to the controller will kick in.
So it may still be a new introduced bug, but also some kind of setting indicating it may need to send >1 value.

sevelen, 2018-07-13:
Thanks i did a temporary change in the code before the relese is up to check solution. It looks working good

TD-er, 2018-07-13:
Great, then I will merge it now.

closed 2018-07-13 by TD-er