letscontrolit/ESPEasy #1563: 8285 locked after set device

puppetmaster886, 2018-07-12:
### Summarize of the problem/feature request

I flash the sonoff 4CH (esp8285) with the build "normal_ESP8285_1024" using atom, it works great.
When I set the device AM2301 on the GPIO08 and enable it. then it stops responding: there is no wifi, it does not respond by serial, it does not respond to reset button and finally I try to flash it again.
It says that it was successfully flashed but it still does not respond.
This same behavior happened 2 times, on 2 different sonoff devices that I tested.

### Expected behavior

it should continue working or go back to factory defaults

### Actual behavior

I do not do anything. 
The only reason that I do NOT think it is bricked is that I can successfully flash it, but I do no respond anyway

### Steps to reproduce

1. flash normal_ESP8285_1024
2. set device AM2301 to GPIO08
3. try to access web page
4. try to flash it again

### System configuration

Hardware:
Sonoff 4CH r1
https://www.itead.cc/wiki/Sonoff_4CH

ESP Easy version: 
mega-20180625

```

```

TD-er, 2018-07-12:
[Schematic 4ch](https://www.itead.cc/wiki/images/d/d3/Sonoff_4CH.SCHMATIC.pdf)
It is connected to the TEM2 line, which is connected to the jack plug.
[Here](https://github.com/arendst/Sonoff-Tasmota/wiki/Expanding-4CH) is stated that these pins cannot be used, since they are internally used for flashing memory.

uzi18, 2018-07-12:
If we can force user to set mcu module type we can avoid to set improper configuration for some gpios =)

TD-er, 2018-07-12:
The problem is I cannot even detect the difference between an ESP8266 or an ESP8285.  (although thinking of it now, I could maybe look at the flash vendor ID)

uzi18, 2018-07-12:
It is better to leave it for user to choose hardware type, but we need to force this question on every module on fresh configuration

TD-er, 2018-07-12:
Fully agree, we should have some default settings options for a number of standard devices with some known to work setup.

closed 2022-11-09 by TD-er