letscontrolit/ESPEasy #1565: Fresh install on ESP8266 - reboots fault

flexiti, 2018-07-12:
System restarts every few second
(2018-07-12  dev_ESP8266_4096)
```
INIT : Booting version:  (ESP82xx Core 2_4_1, NONOS SDK 2.2.1(cfd48f3), LWIP: 2.0.3)
90 : INIT : Warm boot #24 - Restart Reason: Power on
93 : FS   : Mounting...
117 : FS   : Mount successful, used 75802 bytes of 957314
124 : CRC  : No program memory checksum found. Check output of crc2.py
130 : CRC  : SecuritySettings CRC   ...OK 
236 : INIT : Free RAM:20336
237 : INIT : I2C
237 : INIT : SPI not enabled
254 : INFO : Plugins: 74 [Normal] [Testing] [Development] (ESP82xx Core 2_4_1, NONOS SDK 2.2.1(cfd48f3), LWIP: 2.0.3)
255 : WIFI : No valid wifi settings
257 : WIFI : Could not connect to AP!
260 : WIFI : Set WiFi to AP
1183 : WIFI : AP Mode ssid will be ESP_Easy_0 with address 192.168.4.1

```

TD-er, 2018-07-12:
What hardware is this and what devices/sensors are connected?

On the Tasmota issue list I have seen an issue about Sonoff nodes not being able to connect to WiFi when powered via grid power. (connection was then possible when connected to the flasher)

So could you add a bit more information, since this log only indicates it cannot find proper wifi settings.

flexiti, 2018-07-12:
Sonoff Basic, nothing conected, only oryginal flash 1M is replace by 4M winbond 25Q32F

TD-er, 2018-07-12:
Have you tried erasing the flash completely before flashing?

uzi18, 2018-07-12:
More complete log could be useful here.

s0170071, 2018-07-13:
> Have  you tried erasing the flash completely before flashing?

@TD-er  Arduino IDE fails with this.
```

Global variables use 46672 bytes (56%) of dynamic memory, leaving 35248 bytes for local variables. Maximum is 81920 bytes.
/home/john/ArduinoPortable/arduino-1.8.5_ESPgit/hardware/esp8266com/esp8266/tools/esptool/esptool -vv -cd nodemcu -cb 921600 -cp /dev/ttyUSB0 -ca 0x0 -cz 0x400000 -ca 0x00000 -cf /tmp/arduino_build_266704/ESPEasy.ino.bin 
esptool v0.4.12 - (c) 2014 Ch. Klippel <ck@atelier-klippel.de>
An error occurred while uploading the sketch
	setting board to nodemcu
	setting baudrate from 115200 to 921600
	setting port from  to /dev/ttyUSB0
	setting address from 0x00000000 to 0x00000000
error: Invalid argument or value after -cz (argument #11)

```

TD-er, 2018-07-13:
Judging the year (2014), my first guess is the version of ESPtool may be a bit too old.

s0170071, 2018-07-13:
Doesn't it come with the core libraries ? Those I have updated like 2 month ago.

PR #2559 (2019-08-26): [WiFi] Improve WiFi reconnect stability and fix initial setup
Simplified the WiFi connect state machine.
Fixed setup of initial WiFi configuration.

AP mode is now kept as long as client is connected.
AP mode will be started immediately if there are no valid WiFi credentials, or after 20 seconds of no wifi connection.
AP mode will be disabled 60 seconds after starting it, or after the last client disconnected from the AP.

Fixes #2561 
Fixes #2338 
Fixes #2203 
Fixes #1987 
Fixes #1737
Fixes #1640 
Fixes #1565
Fixes #1479 
Fixes #1098
Fixes #558
closed 2019-08-26 by TD-er