letscontrolit/ESPEasy #1566: IP blocked

sanderlv, 2018-07-12:
After flashing my sonoff with the latest "Release mega-20180712" the unit reboots and I can shortly connect to the module via the "new" SSID and I get the "login" screen. But then I get "IP Blocked" after clicking somewhere.

TD-er, 2018-07-12:
Do you have any information from the serial log?
You could try @Grovkillen 's flash tool, which sets the wifi credentials right after flashing.

sanderlv, 2018-12-07:
Every version I put on my devices... always "IP blocked:" and then I have to go via serial and do clearaccessblock VERY ANNOYING :-)

TD-er, 2018-12-07:
Since it is a duplicate of [this issue](https://github.com/letscontrolit/ESPEasy/issues/1850) one, I will close this one.

closed 2018-12-07 by TD-er