letscontrolit/ESPEasy #1567: Invalid value detected for controller OpenHAB MQTT

s0170071, 2018-07-12:
Build= #1562
Trying to post the uptime to MQTT fails. These are the settings:

[image]

Log: 

```

INIT : Booting version: (custom) (ESP82xx Core ea4720b0, NONOS SDK 2.2.1(cfd48f3), LWIP: 2.0.3)
94 : INIT : Warm boot #5 - Restart Reason: External System
98 : FS   : Mounting...
123 : FS   : Mount successful, used 75802 bytes of 957314
137 : CRC  : No program memory checksum found. Check output of crc2.py
164 : CRC  : SecuritySettings CRC   ...OK 
268 : INIT : Free RAM:25224
269 : INIT : I2C
269 : INIT : SPI not enabled
270 : System: Error - To much Plugins
817 : INFO : Plugins: 50 [Normal] (ESP82xx Core ea4720b0, NONOS SDK 2.2.1(cfd48f3), LWIP: 2.0.3)
817 : Programming error! - Increase PLUGIN_MAX
819 : WIFI : Set WiFi to STA
853 : WIFI : Connecting MNET attempt #0
971 : LoopStats: shortestLoop: 10000000 longestLoop: 0 avgLoopDuration: 0.00 systemTimerDuration: 0.00 systemTimerCalls: 1 loopCounterMax: 1 loopCounterLast: 1 countFindPluginId: 1
979 : WD   : Uptime 0 ConnectFailures 0 FreeMem 25128
1971 : SYS  : 0.00
3744 : WIFI : Connected! AP: MNET (38:10:D5:37:FF:95) Ch: 6 Duration: 2891 ms
3774 : WIFI : DHCP IP: 192.168.1.4 (Test-4) GW: 192.168.1.1 SN: 255.255.255.0   duration: 28 ms
3788 : Webserver: start
3884 : MQTT : Intentional reconnect
3885 : LoadFromFile: config.dat index: 28672 datasize: 724
3927 : MQTT : Connected to broker with client ID: ESPClient_60:01:94:1D:4A:CD
3929 : Subscribed to: /Test/#
3972 : SYS  : 0.00
3973 : Invalid value detected for controller OpenHAB MQTT
4027 : Current Time Zone:  DST time start: 2018-03-25 02:00:00 offset: 120 minSTD time start: 2018-10-28 03:00:00 offset: 60 min
5970 : SYS  : 0.00
5972 : Invalid value detected for controller OpenHAB MQTT

```

TD-er, 2018-07-12:
Maybe these issues are related?
https://github.com/letscontrolit/ESPEasy/issues/1559#issuecomment-404610439

s0170071, 2018-07-13:
Its related. The problem sits in Contoller.ino function 

```
boolean validUserVar(struct EventStruct *event) {
  for (int i = 0; i < VARS_PER_TASK; ++i) {
    const float f(UserVar[event->BaseVarIndex + i]);
    if (!isValidFloat(f)) return false;
  }
  return true;
}
```

It tries to check all possible varspertask which is 4. 
The controller only uses one, so the check fails if it checks the uservar no. 2 (cause it is not set)
I suggest to use 
```
boolean validUserVar(struct EventStruct *event) {
  byte valueCount = getValueCountFromSensorType(event->sensorType);
  for (int i = 0; i < valueCount; ++i) {
    const float f(UserVar[event->BaseVarIndex + i]);
    if (!isValidFloat(f)) return false;
  }
  return true;
}
```

TD-er, 2018-07-13:
Sounds like a proper change. I will look at GitBlame, to see who changed it (probably me) and when (to see what I was thinking)

s0170071, 2018-07-13:
thanks. That sounds like you save me the PR hustle with GitKraken.
Btw, does the windows version of GitKraken has a minimize/close button? The linux version does not which really sucks !

TD-er, 2018-07-13:
The Linux version also have a minimize/close button, but it may depend on the window manager used if it is working.
A colleague of mine was also experimenting with window managers and some show similar issues like you describe.

This is my workstation at work:
[image]

closed 2018-07-13 by TD-er

8none1, 2020-06-29:
I think this issue might be a duplicate of this: https://github.com/letscontrolit/ESPEasy/issues/2892