letscontrolit/ESPEasy #1571: Feature: firmware self test

s0170071, 2018-07-13:
During development it happens  that basic features fail. I would propose a self test as far as this is possible.  Test scenarios could be: 

1. connect to wifi repeatedly.
2. discard all devices and controller data and use a known setup instead. (just rename and rewrite config files maybe ?)
3. send data (maybe uptime) to ESP itself using a controller in intervals
4. receive self-sent commands.
5. check free mem, cpu load, boot time, task delays. 
6. implant a cpu loading task and check stability (WD must not kick in)
7. Send and receive to MQTT
All that can be done with no hardware attached.

 etc....

Maybe this can be done as a plugin ?

TD-er, 2018-07-13:
I had as idea to allow the option to disable the rules and the active tasks at a reboot-by-crash.
Maybe one-by-one to allow to regain control over the node when settings may cause a crash.

tonhuisman, 2023-09-03:
This seems to be solved, so can be closed.

TD-er, 2023-09-04:
The suggested steps are not all implemented.
But we do have some fallback options present in ESPEasy since.
For example disabling tasks/controllers/rules on bootloops.
Also the codeing style has changed a lot since, where we now even perform quite a lot of sanity checks at compile-time and I regularly run PVS-Studio on our code to help find issues in the code.
Actually PVS-Studio did find quite a few bugs which would have been quite hard to find or reproduce.

Since this issue was raised, I also added the timing stats feature to get some idea of how often functions are called and how long they take to execute.
This has proven to be a very useful tool to find potential issues.

closed 2023-09-04 by TD-er