letscontrolit/ESPEasy #1572: No Wifi after deep sleep is enabled

barthm1, 2018-07-14:
Deep Sleep not enabled - One Controller - Thingspeak

[image]

Deep Sleep enabled 
[image]

No WiFi conenction is made

TD-er, 2018-07-14:
This happens sometimes, or every time?

TD-er, 2018-07-14:
Seems related: https://www.letscontrolit.com/forum/viewtopic.php?p=29460#p29460

barthm1, 2018-07-14:
yes - can reproduce it with different Chips and sensors-happens after the changes in april

TD-er, 2018-07-14:
OK then I will keep this in mind when looking at scheduling tasks in ESPeasy.

barthm1, 2018-07-31:
Even with  20180723 same behaviour

first boot with Deep Sleep enabled, second without - connect to AP is not made

[image]

TD-er, 2018-07-31:
It is much easier to read when you don't include screenshots, but just cut/paste the text.
Also screenshots in JPEG make it also worse to read. When using screenshots, try PNG instead of JPEG, but for logs, plain text is the best.

tonhuisman, 2023-09-03:
This seems to be solved, so can be closed.

closed 2023-09-04 by TD-er