letscontrolit/ESPEasy #1573: Crash loop when enable SSDP

stephjj, 2018-07-14:
When SSDP is enable, the device crash immediately.

Hardware:
Wemos D1 Mini R2

Software :
Build
20102  - Mega
Libraries
ESP82xx Core 2_4_1, NONOS SDK 2.2.1(cfd48f3), LWIP: 2.0.3
GIT version
mega-20180712
Plugins
48 [Normal]
Build time
Jul 12 2018 02:16:23
Binary filename
ESP_Easy_mega-20180712_normal_ESP8266_4096.bin

TD-er, 2018-07-14:
I guess this also is very hard to recover from? (other than clear and start all over)

stephjj, 2018-07-14:
Here the crash log:

```
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v614f7c32
~ld
⸮U96 : 

INIT : Booting version: mega-20180712 (ESP82xx Core 2_4_1, NONOS SDK 2.2.1(cfd48f3), LWIP: 2.0.3)
97 : INIT : Warm boot #53 - Restart Reason: Exception
101 : FS   : Mounting...
126 : FS   : Mount successful, used 75802 bytes of 957314
438 : CRC  : program checksum       ...OK
469 : CRC  : SecuritySettings CRC   ...OK 
575 : INIT : Free RAM:22672
576 : INIT : I2C
576 : INIT : SPI not enabled
593 : INFO : Plugins: 48 [Normal] (ESP82xx Core 2_4_1, NONOS SDK 2.2.1(cfd48f3), LWIP: 2.0.3)
594 : WIFI : Set WiFi to STA
626 : WIFI : Connecting Masaq attempt #0
639 : LoopStats: shortestLoop: 10000000 longestLoop: 0 avgLoopDuration: 0.00 systemTimerDuration: 0.00 systemTimerCalls: 1 loopCounterMax: 1 loopCounterLast: 1 countFindPluginId: 1
647 : WD   : Uptime 0 ConnectFailures 0 FreeMem 21400

Exception (28):
epc1=0x4021199d epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000004 depc=0x00000000

ctx: cont 
sp: 3fff39e0 end: 3fff3c60 offset: 01a0

>>>stack>>>
3fff3b80:  00000002 0000002f 3fff1ee0 402508c3  
3fff3b90:  3ffe923b 00000000 3fff1ee0 0000002a  
3fff3ba0:  3fff5e12 3fff5e13 3fff1de4 40250940  
3fff3bb0:  3ffe0030 3fff2b74 3fff2b74 4024f8f4  
3fff3bc0:  00000002 3fff5de4 3fff1de4 402208e5  
3fff3bd0:  0000002f 00000000 3fff14ec 3fff2c34  
3fff3be0:  3fffdad0 00000000 3fff3c10 3fff2c34  
3fff3bf0:  3fffdad0 00000000 3fff14ec 3fff2c34  
3fff3c00:  3fffdad0 00000000 3fff14ec 4024af4a  
3fff3c10:  3fff5de4 0000003f 0000002f 402177f4  
3fff3c20:  3fffdad0 00000000 3fff137c 4024d0fd  
3fff3c30:  feefeffe feefeffe feefeffe 3fff2c34  
3fff3c40:  3fffdad0 00000000 3fff2c2e 402514ac  
3fff3c50:  feefeffe feefeffe 3fff2c40 40100700  
<<<stack<<<
```

s0170071, 2018-07-14:
duplicate of #1330, only with exception 28

tonhuisman, 2023-09-03:
This issue seems no longer relevant, and can be closed.

TD-er, 2023-09-04:
I've looked at the SSDP code a few times and it for sure is a memory allocation issue as we need to allocate quite large packets of data.
Not sure if we can 'stream' the data as you probably need to know the exact size of the message in advance.