letscontrolit/ESPEasy #1575: After mega-20180606 distance sensor only reports no reading

StefanRu1, 2018-07-14:
Hi,

i have 2 wemos d1 mini with a HC sr04 and the other one with a US 100.
Including firmware mega-20180606 the sensors reports fine.
All later firmwares says no reading.

Do you have an idea whats going wrong?
The sensors are connected to D12 and D13.

Best regards,
Stefan

TD-er, 2018-07-15:
Good to know it was working until 6th of June.
There was already another report about this plugin, if my memory serves me right.

StefanRu1, 2018-07-15:
Hi, 
thanks for your answer.
Yes i can say it works with 20180606 and does not work with 20180708.
I had no versions in between.

If i should test a version feel free to ask me.

Best regards,
Stefan

ShardanX, 2018-07-21:
Former bug report #1518

jayrisko, 2018-07-24:
I can also confirm that since the mega build of 20180606 the HC-SR04 plugin is not working - just returns "0" readings.  Tried with several types of ESP-8266-based modules with 4MB memory with same (bad) return of 0.

Regards,
Jay

DittelHome, 2018-08-09:
100 % agree

DittelHome, 2018-08-16:
Is anybody working on it ?
Are there any news regarding this issue ?
I have testet with the 20180815 release, same situation ..

tparvais, 2018-08-20:
same problem for me with 08082018 source

NorbertRoller, 2018-08-21:
Issue still in the code 21082018.
I check with a Logic  Analyser and the trigger Line stays high after boot. It should only send an impulse of some ms and repeat in the given time.
As the trigger is not correct, the data impulse is not send from the module.
I am using a RCW-0001
20180606 does work

DittelHome, 2018-08-21:
Is it possible to do a regression for this Plugin ?

jayrisko, 2018-09-03:
Any updates on this issue?  The distance sensors are still broken...

Sartog, 2018-09-14:
I have the same problem.
Any updates on this issue ?

jayrisko, 2018-09-24:
Still no fix or updates on this?  It's been broken for quite a while now...   :(
-Jay

JojoS62, 2018-09-26:
I've tried to track this error. It looks like some memory on the heap gets overwritten.
On init, an instance of NewPingESP8266 is created with the correct TriggerPin and stored as a member variable. I have toggled the pin to verify this. Then on the next calls, the member variable _triggerPin is zero and so no triggers are created. Within the NewPingESP8266 class, I can see no dangerous write operations or assignements to _triggerPin. So it looks like the bug is a side effect by some other memorykiller.
These are my first steps in compiling and testing ESPEasy. Why is it not possible to use addLog() in the Lib codes and what can be used instead to get debug outputs?

lhurt, 2018-09-27:
> I've tried to track this error. It looks like some memory on the heap gets overwritten.
> On init, an instance of NewPingESP8266 is created with the correct TriggerPin and stored as a member variable. [...]

Thanks for starting to tackle this issue. As I'm unfortunately not able to help I keep my fingers crossed that you succeed.

marcohald, 2018-09-30:
Hi,

I have tried Version mega-20180930 but i got the same error.
97676 : ULTRASONIC : TaskNr: 10 Distance: No reading!

with mega-20180524 it is working:
32875 : ULTRASONIC : Distance: 3.00
32877 : EVENT: dst#Distance=3.00

Is there anything I can test ?

Grovkillen, 2018-09-30:
Do you have the possibility to move the device to task number 1 to 4 and test?

TD-er, 2018-09-30:
I will re-open this issue to make a test setup myself to see what's happening here.
It was reported to be working now, but since I never used the plugin myself, I cannot see what's wrong here.

JojoS62, 2018-09-30:
@marcohald can you confirm that it is working with Max. Distance = 500 cm?
returning warnings or errors requires changes in the underlying sonar class. The 'ping' does not return or set a status, a reading of 0 can be an error or out of range. I will try to add a status  that can be printed if distance is 0 to keep the sonar class compatible.

TD-er, 2018-09-30:
I am also in favor of setting a proper default.

JojoS62, 2018-09-30:
yes, but 0 could be a proper default and disable the max. distance checking. There is another check for no response that must return an error.

TD-er, 2018-09-30:
A longer max distance will also result into longer read times and perhaps also more false positives.
So maybe 1 meter is a proper default and when needed more, it can be set.
Most users will not change defaults when things just work, so better have the best experience then :)

marcohald, 2018-09-30:
It is working now with 500 as distance.

101006 : ULTRASONIC : TaskNr: 10 Distance: No reading!
106006 : ULTRASONIC : TaskNr: 10 Distance: No reading!
107456 : FILE : Saved config.dat
107516 : FILE : Saved config.dat
107549 : FILE : Saved security.dat
107556 : ULTRASONIC : TaskNr: 10 TrigPin: 13 IRQ_Pin: 12 max dist cm: 500 max echo: 28557 nr_tasks: 1
107657 : ULTRASONIC : micros() test: 100 msec, 100078 usec, 
107993 : ULTRASONIC : TaskNr: 10 Distance: 3.00
107996 : EVENT: dst#Distance=3.00

I thought that 0 is infinite but that is apparently wrong

JojoS62, 2018-09-30:
No problem. I have changed this behaviour now and currently I'm testing new error messages.

JojoS62, 2018-09-30:
The modification look good: 

- I get errors messages now in the log when the trigger or echo pin is not connected (Wiki for this sensor should be extended)
- max. distance = 0: no distance checking, returns max. distance that the sensor can send
- max. distance >0: error message in log, value output is set to 0.0 cm
- in state mode: errors are logged when sensor read fails (but logging freq is 10 per second)

? Log messages like cyclic distance reading are at INFO level. Should this be changed to DEBUG or is this the standard for other modules also?
The NewPingESP8266 has a lot of #defines for various sensors/plattforms. This makes the code hard to read. I have added the error checking also to the unused cases, but I cannot test these.

TD-er, 2018-10-02:
Other changes can be discussed at the related PR: #1831
So for now I will close this issue,

closed 2018-10-02 by TD-er