letscontrolit/ESPEasy #1582: GPIO switching between  Output back to Input Hi-z mode

hacajd, 2018-07-16:
### Summarize of the problem/feature request

YOUR TEXT GOES HERE
Hi fellows, couple of months I realize problem.
After i used GPIO as OUTPUT in rules section for example, I'm not abble  using same pin back as imput for status check.
I realize its plainly said,but I have that problem. 
Example: I using GPIO5 as input for staus check.
Then I set GPIO5 to status 1.
After all ,what in the world I must use for command to switch back to Hi-z input mote to allow check its state.
Hope , I wrote plainly. I realized after I set GPIO to output state Im not abble use it as input,untill reboot ESP. Thanks for reading this. 

### Expected behavior

YOUR TEXT GOES HERE

### Actual behavior

YOUR TEXT GOES HERE

### Steps to reproduce

1. 
2. 
3. 

### System configuration

Hardware:

ESP Easy version: 

ESP Easy settings/screenshots: 

### Rules or log data

```

```

Budman1758, 2018-07-16:
GPIO's in ESPEasy will work as output by default without any configuration. For status check it must be defined as a switch input device. If all you are trying to do as an input is to tell if its 0 or 1 set up a device as a input switch.  See the Wiki.  

https://www.letscontrolit.com/wiki/index.php/Switch

tonhuisman, 2023-09-03:
This seems to be solved, so can be closed.

closed 2023-09-04 by TD-er