letscontrolit/ESPEasy #1585: ESP32 internal touch sensing - feature request

melwinek, 2018-07-18:
It would be useful - in switch plugin.

like this:
https://github.com/espressif/arduino-esp32/issues/112

TD-er, 2018-07-18:
There are several nice features for the ESP32 which are really useful.
For example these touch sensing options and pulse counter.
These will definitely be useful, but it may take a while, since priority now is in getting ESPeasy as stable as possible for a release, which is not yet focussed on ESP32.

melwinek, 2018-07-18:
If ESP32-SOLO-1 will soon be at a price similar to ESP8266, soon we will all change ESP8266 to ESP32 :)

melwinek, 2018-10-25:
@TD-er Is it feasible in the near future? :)
Because this is a function that will help me turn nodes into esp32.

TD-er, 2018-10-25:
> @TD-er Is it feasible in the near future? :)

I think we're approaching "stable" again for ESP8266, so it looks like there will be time again for new features.
Not giving any date, but I think I will start doing new features within one or 2 weeks from now.

melwinek, 2018-10-27:
It looks like it will not do much work with it. Instead of digitalRead you probably need to use touchRead, like this:
http://www.esp32learning.com/code/esp32-capacitive-touch-example.php

sm0k0, 2020-01-09:
Somebody here talking over this challenge ? I tried the last days but i have only poor programming skils so i was not successful...
Would be great if this csn b picked up again.
thanks in advance  & regards
sm0k0

sm0k0, 2020-01-09:
> > @TD-er Is it feasible in the near future? :)
> 
> I think we're approaching "stable" again for ESP8266, so it looks like there will be time again for new features.
> Not giving any date, but I think I will start doing new features within one or 2 weeks from now.

Hi TD-er. Just want to ask if somebody had the time to investigate here. Would be a very nice thing to have.
thanks in advance  & regards
sm0k0

TD-er, 2020-04-25:
See this pull request: https://github.com/letscontrolit/ESPEasy/pull/3034

PR #3034 (2020-04-27): [ADC] Improve oversampling filtering + ESP32 support
Oversampling:
Values only added when not already the min or max value.
This way a number of erroneous min values or max. values (e.g. due to WiFi activity) will not affect the averaging function.

Perform first reading of ADC before any WiFi activity is taking place. This first reading is shared among VCC readings of sysinfo plugin and ADC plugin.

ESP32 improvements:
- Display detailed ADC pin capabilities/description for GPIO selection
- Only allow GPIOs to be selected for pins that can actually be used for ADC.
- Allow to read the interna [... cut]
closed 2020-04-27 by TD-er