letscontrolit/ESPEasy #1588: Using ESPEasy along with the Serial UART Interface

s0170071, 2018-07-19:
# Scope
These notes will cover some aspects of plugin programming for ESPEasy using the serial interface of the ESP8266. Software considerations also apply to the ESP32. 

### Why should I read this? 
If your plugin code does not work or crashes occasionally.

# Hardware

## How many Uarts are there?

The ESP8266 has two full hardware UARTS. 
UART0 is mapped to RX0/TX0, may be swapped to RX2/TX2
UART1 is mapped to pin TX1. RX1 pin is available on the ESP chip (pin23, IO8) usually not routed to the PCB headers (e.g. at the Wemos D1 mini).

## How large is the buffer ?
The chip has 128 byte hardware FIFO for both read and write operations each. As of Arduino Core version 2.4.0rc2 there is also a software buffer. 

## How can I use the serial debug interface (console) and communicate with another device at the same time?
1. Connect your device to RX2/TX2 like shown here: 
https://github.com/bertmelis/VitoWiFi/issues/27
2. Use swap() to route Serial pins from RX0/TX0 to RX2/TX2. Call swap() only once. If you call it again it will swap the pins back.
3. TX0 pin is now unused by the ESP but still connected to the console. Connect TX1 and TX0. Use Serial1 for debug messages. 

Make sure you do not load the TX2 pin; it needs to float during boot.
The method also allows flashing your ESP without disconnecting it from the serial device. 

# Software

## Basics:
The ESPEasy uses multiple plugins which are scheduled in a round robin fashion. This means that each plugin must make sure it does not block the system. 
Serial data from the UART comes in asynchronously. Always assume that there can be any number of bytes more OR less than you expect. Also, assume that they can arrive at any time. In addition, be aware that there may be some time between bytes.
It is therefore very much recommended that you design your code reentrant using a receive state machine. https://en.wikipedia.org/wiki/Reentrancy_(computing)
This means that you do never wait for bytes to come in. If they are available you read them. If there are none your code returns immediately. 
In most cases, it is ok to do just the receiving part in a re-entrant style. Calculations such as check sums or buffer parsing are usually fast compared to receiving bytes.
Be aware that variables which are defined inside the reentrant function are lost if you exit the function. Make sure they are static. (Global or in the plugin object)

## My code only fails after some days. Wtf?
It is VERY good practice to assume ANY kind of failures in the code you write. So implement checks. Do it as fail proof as possible. The reason for that is that modern hardware with its tiny structures suffers from spontaneous bit flips induced by (cosmic) radiation or power fluctuations. These events are rare but they happen. It is just a matter of time.
Do fail tolerant coding, e.g. 
- initialize your variables as often as you can, especially buffer pointers.  
- do not compare i==40 if you can do i<40 instead. i may have jumped to 42 due to a bit flip.
- use the default branch of a switch statement
- do range checks even if you think that this may never go outside the range.
- use timeouts.
- check the available memory.
In short: just put everything that you do know about your function in a code line. Assume that you may receive any number of corrupt bytes at any time. Even during your final checksum calculations...

## Why could there be more bytes than I expect? 
A faulty wiring, especially bad contacts or noise on the wire may trick the UART into believing that it received data that was never sent.

## Why must I not wait for data? 
Any number of hardware failures may interrupt your communication. There is no flow control nor is there a hardware checksum. If you still wait for data to appear you block the system and ultimately trigger the watchdog.

## Where is the documentation of the Arduino core library? 
http://esp8266.github.io/Arduino/versions/2.3.0/doc/reference.html#serial
and
https://github.com/esp8266/Arduino/blob/21630803626cc88b0af9f82e82dbaf91dc9b2fbb/doc/reference.rst

## RX Buffer
As of core version 2.4.0 rc2 there is also a software receive buffer. It is initialized to 256 bytes. 
For reasons of backwards compatibility the available() function always returns 128 if there are 128bytes or more in the software RX buffer.

## How do I change the RX software buffer size?
Use setRxBufferSize(size) after begin()

## How do I use a software RX buffer?
It is enabled by default. If you set its size to zero, it is disabled. 

## Initialization:
What’s first ? begin() or swap() ? Always do begin() first, any other settings after that.

## How do I detect a receive overrun?
If you only use the hardware fifo (->setRxBufferSize(0)) an overrun is indicated if available() returns 128.
If you use the software buffer, there is no way to detect an overrun condition as the software buffer will always return 128 bytes available if there are 128 or more available.
There is also no way to detect how many bytes exactly are stored in the software buffer. 

## How do I handle an overrun condition? 
If you have an overrun that means that you have missed some bytes. In most cases, the remaining data is either, incomplete or outdated so just discard it.

## How do I empty the receive buffer? 
read() until available() returns 0.

## How long does receiving take
That depends on your baud rate. At 9600 bps a character requires about 1ms. 
Reading data that is already available and storing it into ram takes only a few microseconds. 

## How many bytes can I receive? 
As many as you like, BUT make sure that you 
1. Do not block the system, i.e. use a timeout in your function. Go for about 10ms timeout.
2. Do not use for loops for receiving (see 1.).
3. Always check your buffer pointer right before writing.

## I keep missing bytes. What’s wrong? 
Here is a small checklist: 
1. Check your wiring and try a different cable. If you own an oscilloscope, use it.
2. Check that the ESPEasy scheduler has called your code in time. Other plugins or network timeouts may have delayed it. 
3. Check for buffer overruns if you do not use the software buffer.
4. Are there spontaneous reboots or watchdog timeouts?
5. Do you use the correct baud rate setting?

## Shoud I use PLUGIN_TEN_PER_SECOND or PLUGIN_SERIAL_IN ? 
Depends. 
PLUGIN_SERIAL_IN is invoked once each task has been run and serial data is available.
PLUGIN_TEN_PER_SECOND is invoked even if there is no data available. Same goes for PLUGIN_FIFTY_PER_SECOND. 
All three should do if your code does not block.

## How do I use a receive timeout ?
basicly you only receive as long as data is available and for no more like 10ms. Like that:

unsigned long startTime = millis();
...
while (Serial.available()) {
	if (bufferPtr < BSIZE) 
		buf[bufferPtr++] = Serial.read(); 
	if (millis() - startTime > 10) return;
	}

## Do I have to use yield() or delay() ?
No. Never use delay when receiving data. It blocks the system. 
There is, however, a condition in which the number of available bytes is not updated if bytes are coming in while you execute your plugin. You may use yield() to allow for an update. The better solution is to just let it go and trust in the reentrant capabilities of your function. Next time your plugin executes, the bytes will show. 

## My ESP does not start if I receive serial data.
There is a bug in core version 2.4.0 that crashes the ESP if serial data was received before begin() was executed. It is fixed as of April2018. Use 2.3.0 or the git version of the core or make sure you do not receive data during start up.  https://github.com/esp8266/Arduino/issues/3869

TD-er, 2018-07-19:
Only remark:
Set the timeout time to `unsigned long timeout = millis() + 10;` and call `timeoutReached(timeout)`

And my plan is to have the main scheduler looking into reading data from serial and use the appropriate buffers to fill. Then the plugins can have a much simpler job processing data.

And I wish all "issues" were this elaborate and informative :+1:

tonhuisman, 2023-09-03:
This seems to be solved, so can be closed.

TD-er, 2023-09-04:
Just as a remark before closing this issue.

I re-read the issue again and it is good to know I didn't miss any of the mentioned points in the fixes/changes made in the past few years.
A really well written  summary of the serial ports on ESP processors and how to use them.

N.B. I had forgotten about pin 8 as RX pin for the ESP8266 Serial1 port (or may never have known?) So it was useful to read it again.

closed 2023-09-04 by TD-er