letscontrolit/ESPEasy #1593: [PinState] Make single function for setting state/mode and keeping state

TD-er, 2018-07-19:
Current implementation has `setPinState` to keep track of the mode and value.
This function is called often when the mode or value has changed.
Call this state update from the function that sets the mode/value to make sure no update can be missed.

giig1967g, 2018-10-30:
A function that needs to be optimized is the getPinStateJSON that does a loop of 32 times after every GPIO command in PLUGIN_WRITE.
It is called most of time from the SendStatus function. i.e.:
`SendStatus(event->Source, getPinStateJSON(SEARCH_PIN_STATE, PLUGIN_ID_019, event->Par1, log, 0));`

But considering what the function does, it's not used in rules commands:

```
void SendStatus(byte source, String status)
{
  switch(source)
  {
    case VALUE_SOURCE_HTTP:
    case VALUE_SOURCE_WEB_FRONTEND:
      if (printToWeb)
        printWebString += status;
      break;
    case VALUE_SOURCE_MQTT:
      MQTTStatus(status);
      break;
    case VALUE_SOURCE_SERIAL:
      Serial.println(status);
      break;
  }
}

```
To avoid changing the current logic, I suggest at least to change as follows:

From:
`SendStatus(event->Source, getPinStateJSON(SEARCH_PIN_STATE, PLUGIN_ID_019, event->Par1, log, 0));`

To:
`if (event->Source==VALUE_SOURCE_HTTP || event->Source==VALUE_SOURCE_SERIAL || event->Source==VALUE_SOURCE_MQTT || event->Source==VALUE_SOURCE_WEB_FRONTEND)
SendStatus(event->Source, getPinStateJSON(SEARCH_PIN_STATE, PLUGIN_ID_019, event->Par1, log, 0));`

what do you think?
Should I proceed?

TD-er, 2018-10-30:
My first remark when seeing such a line with lots of OR's comparing to static values is: why not use a switch statement?
It will probably compile to faster code and is better readable.

```c++
switch (event->Source) {
  case VALUE_SOURCE_HTTP:
  case VALUE_SOURCE_SERIAL:
  case VALUE_SOURCE_MQTT:
  case VALUE_SOURCE_WEB_FRONTEND:
    SendStatus(event->Source, getPinStateJSON(SEARCH_PIN_STATE, PLUGIN_ID_019, event->Par1, log, 0));
	break; 
}
```

And I totally agree on the inefficiency of the current implementation.
But I guess I have to see it tomorrow morning again, after some sleep.

giig1967g, 2018-10-30:
of course your design is much better.
I counted 27 matches in the code...

TD-er, 2018-10-30:
27 matches for what exactly?

giig1967g, 2018-10-30:
of
` SendStatus(event->Source, getPinStateJSON(SEARCH_PIN_STATE,....`
that needs to be wrapped around the switch statement.

giig1967g, 2018-10-31:
@TD-er just tell me if you want me to code the change

giig1967g, 2018-11-07:
@TD-er should I create a PR for this?

TD-er, 2018-11-07:
Please do, but if possible try to wait **pushing** a PR until I got the PUYA patch merged and verified (caches and such)

giig1967g, 2018-11-07:
ok

tonhuisman, 2023-09-03:
This seems to be solved, so can be closed.

closed 2023-09-04 by TD-er