letscontrolit/ESPEasy #1595: Md5 check	fail ! on system info page

flexiti, 2018-07-20:
Maybe nothing important, but I do not like  :) as something is shown in red. Can not do anything about it? :)

TD-er, 2018-07-20:
If it is your own build, then the MD5 check will fail. :)
Or is it about the settings?
Then I may need to change something to get that MD5 stuff working again. (I disabled it a while back due to issues when changing between versions)

flexiti, 2018-07-20:
Yes,its my own build.
Maybe just blind it until it's done? ... because it does not work anyway

TD-er, 2018-07-20:
We should remove the checksum when the crc-script was not run (for people building their own)

clumsy-stefan, 2018-07-20:
or run the crc-script after building (which is what I do) to have a correct checksum!

s0170071, 2018-07-20:
... use the crc2.py md5 generator which is in the ESPEasy repo. 
Installation instructions here: https://github.com/s0170071/CRC4ESP

vmfs1968, 2018-07-20:
crc2.py does not work on esp32 does anyone know why?

s0170071, 2018-07-21:
...was never built for and  tested with ESP32

Grovkillen, 2019-10-10:
I close this one

closed 2019-10-10 by Grovkillen