letscontrolit/ESPEasy #1597: [RTC memory] Rethink what needs to be stored in RTC to survive reboot and sleep

TD-er, 2018-07-20:
Currently the functions `saveUserVarToRTC()` and `readFromRTC()` handle storing last known variables of the plugins to RTC

`saveToRTC()` and `readFromRTC()` handle the storage of the `RTCStruct`:

```C++
// this offsets are in blocks, bytes = blocks * 4
#define RTC_BASE_STRUCT 64
#define RTC_BASE_USERVAR 74

//max 40 bytes: ( 74 - 64 ) * 4
struct RTCStruct
{
  byte ID1;
  byte ID2;
  boolean unused1;
  byte factoryResetCounter;
  byte deepSleepState;
  byte unused2;
  byte flashDayCounter;
  unsigned long flashCounter;
  unsigned long bootCounter;
} RTC;
```

This struct has no proper constructor, so it could contain any information at boot.
Also, since the RTC storage is very sparse, the layout of this data should be done different.
Start with the biggest items (unsigned long) and add smaller ones.
Current implementation is a waste of space.

Also not all active tasks/plugins must have their values stored.
Often the previous value is only useful when using a formula and not all plugins have 4 values.

Also consider storing last WiFi connection information, like BSSID and channel. That information could help improving WiFi connection speed.
Maybe also store last used IP config? (DHCP may take a few seconds to complete)

Other data could contain information on last changed setting, or whether to boot in safe mode at some crash. (first set next boot mode to safe mode when loading settings and after successful load, set it to normal mode)

TD-er, 2018-07-20:
I have looked into getting more info about the DHCP request (also to get NTP server suggestion and a hostname), but I did not find a way to get access to that info.

But when setting deep sleep, and not waken by some external trigger, then you could store the last timestamp before sleep and thus have a reasonable good idea of the current time when waking from the sleep.

TD-er, 2018-07-29:
There is already a work-around present in the code to get out of boot loops:

```C++
/********************************************************************************************\
  If RX and TX tied together, perform emergency reset to get the system out of boot loops
  \*********************************************************************************************/
void emergencyReset()
```

Meaning, when both of these pins are tied together, it will try a factory reset.

uzi18, 2018-07-29:
so we not need any other pin to emergency reset :)