letscontrolit/ESPEasy #1604: ESP32 not boot with latest firmware

melwinek, 2018-07-22:
I bought ESP32 for testing.
For flashing i use flashSerial.cmd from ESPEasy32_R20100.zip.
ESP32 boot with: ESPEasy32_R20100
ESP32 not boot with: ESPEasy_mega-20180722
log:
```
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:812
load:0x40078000,len:0
load:0x40078000,len:11404
entry 0x40078a28
▒U40 :

INIT : Booting version:  (ESP32 SDK v3.1-dev-239-g1c3dd23f-dirty)
41 : INIT : Cold Boot - Restart Reason: CPU0: Vbat power on reset CPU1: for APP CPU, reseted by PRO CPU
46 : FS   : Mounting...
86 : CRC  : No program memory checksum found. Check output of crc2.py
88 : FS   : Error while reading/writing /config.dat in 1074

PID:0
Version:0
INIT : Incorrect PID or version!
RESET: Resetting factory defaults...
RESET: Cold boot
RESET: formatting...
RESET: formatting done...
2284 : FS   : Error while reading/writing /config.dat in 979
2405 : FS   : Error while reading/writing /config.dat in 1074
2410 : FS   : Error while reading/writing /config.dat in 1009
RESET: Succesful, rebooting. (you might need to press the reset button if you've justed flashed the firmware)
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:812
load:0x40078000,len:0
load:0x40078000,len:11404
entry 0x40078a28
▒U35 :

INIT : Booting version:  (ESP32 SDK v3.1-dev-239-g1c3dd23f-dirty)
35 : INIT : Cold Boot - Restart Reason: CPU0: Software reset CPU CPU1: Software reset CPU
40 : FS   : Mounting...
92 : CRC  : No program memory checksum found. Check output of crc2.py
94 : FS   : Error while reading/writing /config.dat in 1074

PID:0
Version:0
INIT : Incorrect PID or version!
RESET: Resetting factory defaults...
RESET: Cold boot
RESET: formatting...
RESET: formatting done...
2311 : FS   : Error while reading/writing /config.dat in 979
2410 : FS   : Error while reading/writing /config.dat in 1074
2417 : FS   : Error while reading/writing /config.dat in 1009
RESET: Succesful, rebooting. (you might need to press the reset button if you've justed flashed the firmware)
ets Jun  8 2016 00:22:57

```

TD-er, 2018-07-22:
Could you try to flash the [build](https://www.dropbox.com/s/agm7cmgis6vouuj/%5BparseString%5D%20Some%20fixes%20and%20checks%20added%20AND%20quotes%20in%20parameters%20possible%20%231603.rar?dl=0) I just made?

I know for a fact, that one is working, since it is running on my ESP32 module

melwinek, 2018-07-22:
No change.
My module is Wemos with ESP32S

TD-er, 2018-07-22:
Can you use the flash tool to identify the board?
I know Espressif is working on a single-core variant, but it is not yet present in the [list on Wikipedia](https://en.wikipedia.org/wiki/ESP32#Surface-mount_module_boards).
The restart reason is reporting 2 CPU cores.
I will add some check for the flash chip vendor ID too, just to make sure it is no PUYA chip.

melwinek, 2018-07-22:
This is dual core.
flash vendor:
C8h : GD
flash devID:
4016h
QUAD;32Mbit
crystal:
40 Mhz

TD-er, 2018-07-22:
Is it also possible to get an idea on the chip model? (ESP32 chip)

melwinek, 2018-07-26:
After erasing flash with esptool.py, and after flashing ESPEasy_mega-20180722 with flashSerial.cmd all is ok.
You can close issue.

TD-er, 2018-07-26:
Great, glad it has been solved :)

closed 2018-07-26 by TD-er