letscontrolit/ESPEasy #1607: DEFAULT_SSID not set

stefbo27, 2018-07-23:
### Summarize of the problem/feature request

After the first flashing of the ESP the DEFAULT_SSID, DEFAULT_KEY and DEFAULT_NAME (maybe other settings as well) are not set, as they are defined either in ESPEasy-Globals.h or Custom.h. Therefore the device can not connect to WIFI and goes to AP-mode

### Expected behavior

Device should connect to WIFI network

### Actual behavior

YOUR TEXT GOES HERE

### Steps to reproduce

1. Define DEFAULT_SSID etc. in either Custom.h or ESPEasy-Globals.h 
2. Flash current version (mega-20180722) on a NEW chip 
3. show settings (via serial)

### temporary solution
issue reset command via serial, then values from DEFAULT_SSID are set. 
A quick search for DEFAULT_SSID showed that this definition is only used in void ResetFactory(void) in Misc.ino

### Related issue
SSID2 can not be defined in Custom.h

TD-er, 2018-07-23:
Just to make sure, but you built your own version?

This value is written to the default settings when you reset the ESP to factory default.
If you did a clean flash and then reboot after it tries to create an accesspoint, does it connect then?
And what happens when you type "Reset" in the serial terminal?

It sounds like the default settings may not be activated in your setup.

stefbo27, 2018-07-23:
Yes I built my own version.
Usually I have older devices and just re-flash. Then the old settings remain. However this time I flashed the code to a brand-new D1-mini. 
Yes I agree that the defaults for WIFI are not activated. I double checked that in  current code DEFAULT_SSID is only used in reset command (see above)

After issuing "Reset" in serial terminal all works well (as expected when ResetFactory(void) is called). I am not sure If ResetFactory(void) should be called after each flash, or only the settings should be set in the appropriate location.

TD-er, 2018-07-24:
I guess the reset should be called when ESPeasy does not find a proper SPIFF and thus starts to format it.
That would be a great time to call this reset function and read the settings it just wrote.

tonhuisman, 2022-04-18:
This seems to have been fixed. Should probably be closed.

closed 2022-04-18 by TD-er