letscontrolit/ESPEasy #1609: Switch - HC-SR501 Does not work

AxelMilb, 2018-07-24:
Hello
All versions from ESPEasy_mega-20180722 do not switch anymore (Switch - HC-SR501) it stays permanently at 0.
Version ESPEasy_mega-20180720 works.
Sorry for my bad english.

TD-er, 2018-07-24:
Can you post a bit more information like screenshot of the switch setting page, rules you use and maybe some logs if possible?

AxelMilb, 2018-07-24:
I hope that these data will suffice.
The picture with the switch on 1 is from the ESP_Easy_mega-20180720_dev_ESP8266_4096.bin.
Thank you for your efforts.

[image]

[image]

[Log.txt](https://github.com/letscontrolit/ESPEasy/files/2225159/Log.txt)

Budman1758, 2018-07-24:
Not real positive here but seems to me you would NOT want the internal pullup enabled.

TD-er, 2018-07-24:
@Budman1758 : It depends on how the sensor is connected.
And simply said, if it was working before, I don't see an obvious reason why the internal pull-up should not be enabled.

@AxelMilb:
Handling of sensor readings has been improved, so one thing I can think of is that it may detect small noise spikes better than before.
I just tested the switch here on my node (flash button on NodeMCU) and it works flawless.

Could you also try to move the switch plugin to one of the first 3 or 4 task positions?
I did not yet remove the handling limit of "timing-critical" plugins. There is currently still a limit of only the first few task positions to handle those plugins.

TD-er, 2018-07-24:
Maybe this one is related? https://github.com/letscontrolit/ESPEasy/issues/1265

AxelMilb, 2018-07-25:
You're right, if I put the switch in the first task positions he works.
Thanks for your help.

TD-er, 2018-07-25:
OK, so that's one thing I will have to take care of in my timing quest on the ESPeasy :)

closed 2018-07-25 by TD-er