letscontrolit/ESPEasy #1610: MQTT : Intentional reconnect using OpenHAB MQTT

andrea-daru, 2018-07-24:
Hi all,
I'm having problem publishing sensor data on _MQTT_.
I'm using _OpenHAB MQTT_ protocol.
Server side I've installed _Mosquitto_ as MQTT server.
It stopped working since i've updated the firmware to the latest build.
The log shows a lot of `MQTT : Intentional reconnect ` messages.

Anyone else have the same problem? Solution?

**Build information**:
Build | 20102  - Mega
-- | --
Libraries | ESP82xx Core 2_4_1, NONOS SDK 2.2.1(cfd48f3), LWIP: 2.0.3
GIT version | mega-20180723
Plugins | 48 [Normal]
Build Md5 | 29b082ac972b209714adaf5b79e33
Md5 check | passed.
Build time | Jul 23 2018 02:13:36
Binary filename | ESP_Easy_mega-20180723_normal_ESP8266_4096.bin

TD-er, 2018-07-25:
Is the wifi connection stable?

Grovkillen, 2018-07-25:
I leave this issue open but it seems to be a forum level support issue. (I'm currently at my parents place up north of Sweden and will not be able to test the issue myself).

andrea-daru, 2018-07-25:
@TD-er the wifi connection it's very stable... no loss of signal for months!

Grovkillen, 2018-07-25:
@andrea-daru If you re-submit the controller settings, does this still happen?

Grovkillen, 2018-07-25:
My build: 

Build
20102  - Mega
Libraries
ESP82xx Core 2_4_1, NONOS SDK 2.2.1(cfd48f3), LWIP: 2.0.3
GIT version
mega-20180723
Plugins
48 [Normal]
Build time
Jul 23 2018 02:13:36
Binary filename
ESP_Easy_mega-20180723_normal_ESP8266_4096.bin

Causes no issues.

andrea-daru, 2018-07-25:
@Grovkillen yes it happen also if a re-submit controller settings.
We've the same build with different problem... maybe the problem it's somewhere else.
Which version of mosquitto you have installed?

TD-er, 2018-07-25:
Also try connecting on IP base instead of hostname.

One other thing that may trigger this, is the timeout setting on the Mosquito side.
A connection is considered lost (from broker's perspective) when a node does not send a keep-alive message within this timeout. (or 50% longer than this setting)
From memory, I would say the current timeout used in ESPeasy is 10 seconds.
I guess it would not hurt if we made this a user setting. (let's hope it is not a static define at compile time in pubsubclient)

See also: https://github.com/knolleary/pubsubclient/issues/239

Grovkillen, 2018-07-25:
Yes I connect using IP, not hostname.

andrea-daru, 2018-07-25:
Setting the IP instead of the Hostname doesn't work.
The log shows:
```
331873: WD : Uptime 6 ConnectFailures 0 FreeMem 17552
351875: MQTT : Intentional reconnect
351906: MQTT : Failed to connect to broker
361873: WD : Uptime 6 ConnectFailures 0 FreeMem 17552
381975: MQTT : Intentional reconnect
382010: MQTT : Failed to connect to broker
391873: WD : Uptime 7 ConnectFailures 0 FreeMem 17552
```
In the Info page (tools > info) I see a strange "Client IP" that I don't know what is and how I can configure it:

Wifi | 802.11N (RSSI -65 dB)
-- | --
IP config | Static
IP / subnet | 192.168.1.203 / 255.255.255.0
GW | 192.168.1.1
**Client IP** | **192.168.1.15**
DNS | 8.8.8.8 / 0.0.0.0
Allowed IP Range | All Allowed
Serial Port available: | 1 (128 , 0)
STA MAC | EC:FA:BC:8B:9E:02
AP MAC | EE:FA:BC:8B:9E:02
SSID | My FRITZ!Box
Channel | 1
Connected | 5 m 16 s
Last Disconnect Reason | (1) Unspecified
Number reconnects | 0

Grovkillen, 2018-07-25:
So you don't belong to that sub net? I guess that's the problem then.. could you try to fix the issue with the Fritzbox then? I think that's the problem if you don't recognize the IP. I close this for now. I'll reopen it if we find that it is indeed the FW that is causing this.

Grovkillen, 2018-07-25:
PS. client IP is your browsers/phone/computers IP address

TD-er, 2018-07-25:
The client IP is your own IP (you are the client).
Judging from the logs, it cannot connect to the broker.
What is the IP address of the broker?

And as a test, can you set the IP config to "DHCP"?
Since you appear to have a Fritz box, you can login to the Fritzbox first and check the settings in there to make sure the ESP node gets the same IP also when using DHCP. ("Assign same IP" or something like that)

TD-er, 2018-07-25:
Just another thought; Do you have > 1 MQTT controller active on the same ESP node?

andrea-daru, 2018-07-25:
@TD-er Yes I've only 1 MQTT Controller.
I've found the problem... I was setting the wrong IP Addr for the Broker... so it was my fault.
Anyway the resolution of the hostname stopped working... so **it's better to set a static ip to the MQTT broker**.
Thank you all for the support! This project it's amazing :)

closed 2018-07-25 by andrea-daru

andrea-daru, 2018-07-25:
I've found the solution to the problem and it was the DNS IP.
For some reason during the update of the firmware I've lost the DNS setting (I've found the Google DNS 8.8.8.8 instead of my own DNS).
That's why... too simple to be considered 🤣