letscontrolit/ESPEasy #1616: [Boot loops] Handle boot loops with more elegance

TD-er, 2018-07-29:
It should be possible to detect multiple crashes and instead of driving the user nuts to gain access, it would be nice to allow to temporarily disable all plugins, controllers, etc. to regain access.

There is already an option to short RX/TX, which will perform a factory reset, but then you loose all settings.

The last boot reason can be stored in the RTC memory, along with some counter.
Disabling settings can also be done in stages, like first plugin, then controller, etc.
Maybe it is also possible to keep track of the last started action (and remove that from RTC when done)
This could be done from the job scheduler. (id is 32 bit, which is the same size as a RTC slot)

uzi18, 2018-07-29:
We can store boot timestamp in rtc, so we can compare it at next boot ...

TD-er, 2018-07-29:
But what if it crashes before connecting to NTP?
And I have seen strange things regarding the `millis()` counter surviving a (warm) reboot.

TD-er, 2018-07-30:
Also the serial port should then be (re)enabled when recovering from a crash.

But this interactive way sounds great.

s0170071, 2018-07-30:
I think the serial is used already from boot until the point when settings are read.

tonhuisman, 2022-05-07:
This seems to be implemented, so issue can be closed?

closed 2022-05-07 by TD-er