letscontrolit/ESPEasy #1617: [ser2net] Adapt plugin to use new SerialReader

TD-er, 2018-07-29:
Apparently this plugin may lead to crashes, as mentioned on the [wiki](https://www.letscontrolit.com/wiki/index.php/Ser2Net) and reported on the [forum](https://www.letscontrolit.com/forum/viewtopic.php?p=30016#p30016)

The new (not yet finished) SerialReader should take care of this.

KennieNL, 2019-01-22:
This is fixed with mega-20190106 right?
I'm running mega-20190116 and don't face any issues anymore.

TD-er, 2019-01-22:
It should be.
Glad to see it is apparently working now.

Since you're apparently running  this plugin.
Can you also test this PR: https://github.com/letscontrolit/ESPEasy/pull/2235
If you need a test build, I can make one for you.

KennieNL, 2019-01-22:
Sure, send me the test build and let me know what information you need

Im using the P1 plugin, but guess its the as the ser2net?

TD-er, 2019-01-25:
That should not have changed, since it is a representation of how the data is stored in the settings file.
But is it working with either the P1 plugin or ser2net?

KennieNL, 2019-01-26:
Ah no, it breaks the P1 plugin giving no more serial output at all.

TD-er, 2019-01-26:
OK, good to know.

dynasticorpheus, 2019-02-21:
@TD-er  Happy to help and do some testing hence wondering if there is a build I can try? Unfortunately I can only do 1024 build due to HW limitations.

TD-er, 2019-02-21:
It should be merged already, so if you take the last build, then it should be included.
But I'm afraid I may already know the answer of that test....
See https://github.com/letscontrolit/ESPEasy/issues/2343

TD-er, 2019-02-25:
In #2352 I have reverted the code of both P020_Ser2Net and the P044_P1WifiGateway.
So it should now work like it did in build 20181231.
Can you please verify that and close this issue if it is working again?

dynasticorpheus, 2019-02-27:
Installed mega-20190227 and also facing #2356 but admin as user works though. Will report back later on stability with regards to  P1 Wifi Gateway

closed 2019-10-10 by TD-er