letscontrolit/ESPEasy #1620: [GPIO] Make ESPeasy core GPIO handler

TD-er, 2018-07-30:
Add a class for handling watching GPIO pins for changes (input) to set triggers and administer timing statistics (e.g. time up and time down)

Currently there is a rather complex system in P001 (Switch) to sample the input on a GPIO pin.
This should be done in ESPeasy core and the P001 should just be called then to see if something should happen (e.g. send to controller) and a rule event must be sent (add event handling job to the scheduler)

For ESP8266, see [ESP8266 - Technical Reference](https://www.espressif.com/sites/default/files/documentation/esp8266-technical_reference_en.pdf) (starting page 7) and [ESP8266-wiki - GPIO registers](https://github.com/esp8266/esp8266-wiki/wiki/gpio-registers)
For ESP32, see [GPIO & RTC GPIO - ESP-IDF Programming Guide](http://esp-idf.readthedocs.io/en/latest/api-reference/peripherals/gpio.html)

This class should offer an interface to either register a call-back function on a trigger, or just register the last event for that pin and let the `loop()` handle processing those events.
When no interrupt is possible, it should implement its functionality.

Also a realtime 'peek' of all GPIOs must be possible, to avoid caching issues.
Such a peek may be implemented using special registers to read the state of all pins at once.

TD-er, 2018-07-31:
See @clumsy-stefan comment here: https://github.com/letscontrolit/ESPEasy/pull/1618#issuecomment-409093080
GPIO monitoring should be improved (a lot).

workgroupengineering, 2018-07-31:
In scant spare time i'm working to unified gpio of core and extended io (PCA9685, etc.). 
I have thought of something like this:

```c++

enum GPIO_DigitalValue 
{  
  LOW,
  HIGH,
  HIGH-Z,
};

typedef void (*GPIO_WriteAlanog) (byte pin, short value);
typedef void (*GPIO_WriteDigital) (byte pin,enum GPIO_DigitalValue value);

typedef short (*GPIO_ReadAlanog) (byte pin);
typedef enum GPIO_DigitalValue (*GPIO_ReadDigital) (byte pin);

typedef struct GPIODeviceStruct
{
  GPIODeviceStruct(byte minPin, byte maxPin,
   GPIO_WriteAlanog writeAlanog,
   GPIO_WriteDigital writeDigital,
   GPIO_ReadAlanog readAlanog,
   GPIO_ReadDigital readDigital,
   ):
    MinPin(minPin),
    MaxPin(maxPin),
    WriteAlanog(writeAlanog),
    WriteDigital(writeDigital),
    ReadAlanog(readAlanog),
    ReadDigital(readDigital)
  {

  }  
  public:
  const byte MinPin;
  const byte MaxPin;
  const GPIO_WriteAlanog WriteAlanog;
  const GPIO_WriteDigital WriteDigital;
  const GPIO_ReadAlanog ReadAlanog;
  const GPIO_ReadDigital ReadDigital;  
}GPIO_Device;

void Command_GPIO(GPIO_Device device, byte pin, enum GPIO_DigitalValue value)
{

}

void Command_PWM(GPIO_Device device, byte pin, short value)
{

}

void Command_Pulse(GPIO_Device device, byte pin, enum GPIO_DigitalValue initialState,short duration,short repetion)
{

}

void Command_Tone(GPIO_Device device, byte pin, unsigned int tone, unsigned long duration)
{

}

void Command_Rtttl(GPIO_Device device, byte pin, const char *notes)
{

}

```

TD-er, 2018-07-31:
Please have a look at the links I added in the topic.
These allow to read/write special registers, which do just like that.
There is a register which contains the GPIO pin state and one to set a pin value ("1" means change GPIO to 1, "0" means leave the rest like it is)
Also to clear.
And there are also commands to set triggers etc.

I really like to have 1 interface to control them all and manage all and allow interfaces to set specific pin states/modes.
Not structs per GPIO.
This to be able to run the check/update in as few commands as possible to be able to run them as often as possible to get the timings right.
Using the core commands to handle single GPIO pins are just too slow and you can monitor them all at the same time in a single command.

LeeNX, 2018-07-31:
This sounds awesome, but as a none tech/programmer, I have been worrying about pulse counting for things like meters/flow sensors. I am always thinking I might be missing pulses and have been thinking about using triggers. I see the note about call-back function for triggers. I am wondering if there might be a way to list the minimum delay between triggers? Thou a simple flow sensor might not really have a minimum delay, like an electrical pulse meter ... Maybe I am just speaking none sense.

TD-er, 2018-07-31:
Flow sensors should behave quite nicely.
It is the push buttons that may give a lot of pulses when pressed and thus require filtering.

ESP32 has hardware instructions to handle these. ESP8266 must do it in software.

LeeNX, 2018-07-31:
@TD-er - Thanks for all the awesome work!!!  
  
ESP32 is great, but currently overkill for my little projects, where ESP8266 seems to work fine, again, with all the hard work of the ESPEasy community!!  
  
As for flow sensors, can I ask for confirmation or clarification on behave well? Again, my concern about missing pulse, thou as I type this, I am thinking if we used triggers, it might flood the ESP8266 if flow is maxed out.  
  
Basically thinking about how electric meters have impulses per kWh, where a meter will have a max rating of kWh, else it will melt. So you can work out delay between pulses, which I have seen been reported as low as 80ms between pulses.  
  
Am I being overly paranoid?

TD-er, 2018-07-31:
Interval of 80 msec between pulses also means the "up" time of the pulse is low.
That's probably more of an issue than the duration between pulses.
You could add a flipflop between the ESP and the sensor. Such a flipflop will detect very short pulses and make the duty-cycle roughly 50% and also double the pulse interval time. Then it is more likely all pulses will be counted.

TD-er, 2018-08-04:
A task definition is a lot more than just handling what GPIO pin is used.
A task can have up to 4 output values. Each with a name, a number of decimals and a formula.
Each of these can be addressed in rules or other means where string replacements of `[taskname#taskvalue]` can be used.
Also a task has up-to 3 controllers selected, each can have an IDX, etc.
There are also Interval settings and a number of other values.

For ESP8266, there is a default Settings struct for 12 Tasks. For ESP32 this has increased to 32 tasks.
Last week I introduced the graphical view of the settings in the webinterface and it struck me the Settings struct is about 2 kB larger on ESP32. The only change is those 20 extra tasks.
So a rough estimate is a task will take up-to 100 Bytes.

But I totally agree the memory management can be improved at some points :)

clumsy-stefan, 2018-08-04:
Sorry, II think I misled... I didn't meant talking about the memory management I thought more about the ease of use for definitions and servers to use GPIO's. If it's more meaningfull to extend the # of tasks seems to be fine too, but I think extending the core (and the I/O plugins) by a "generic" return value which provides the status of all GIO's at once would also reduce config effort and make it more speedy (less CPU) to transfer the values to the controller/server...

was meant as input as you were thinking about redoing how GPIO's are handled...

the graphical setting is great... in my versions I use 24 tasks, which urn fine so far... so if memory is not the real issue, extending tasks shoul d not be that much of an issue then?

TD-er, 2018-08-04:
With current settings definition, increasing the number of tasks, will render existing settings unusable.

I could add a single define in the code so you can use more tasks if you like. Just a single setting during build.
Just to see what may go wrong.
I will then also swap the order of some settings, to allow to use the same reserved space, but allow to extend it at the end.
With the new graphic view, you can see how it will be arranged.

About the GPIO handling. It is indeed my intention to allow extenders to be included in that same GPIO handler, but it will be done in steps. Maybe even adding the extender GPIOs after the Mega release.

clumsy-stefan, 2018-08-04:
ok.. using 24 tasks work fine with the defines I postet in the other thread. also survived updates like this as long as you keep the same amount of tasks... not sure if it makes sense to preserve settings if #of tasks change...

as a first step you could change the definitions (as I use them) so the user can decide how many tasks he wants to use with the clear warning, that changing the # of tasks between builds/flashes will render old config unusable...

ShardanX, 2018-08-04:
More Ports can be done without more task settings.
See #557

TD-er, 2018-08-04:
I created a new project in this Github, just to be able to track related issues.
So if there are any related issues, please add them here.

clumsy-stefan, 2018-08-04:
cool, thx..

uzi18, 2018-08-07:
idea: split config.dat into 3/4 files...

TD-er, 2018-08-21:
See also reported issue on the forum:
[lost concurrent input events on GPIO pins](https://www.letscontrolit.com/forum/viewtopic.php?f=4&t=5709&p=30589#p30589)

tonhuisman, 2023-09-03:
This seems to be solved, so can be closed.

closed 2023-09-04 by TD-er