letscontrolit/ESPEasy #1621: VCC and BME280 Device Problem on recent builds

TD-er, 2018-07-30:
See [forum topic](https://www.letscontrolit.com/forum/viewtopic.php?f=4&t=5629&p=30050#p30050).

> I've been using the VCC feature to measure battery voltage for my remote battery powered BME280 temperature, humidity and pressure device for a few months. The build that last worked (for me) was the ESP_Easy_mega-20180522_test_ESP8266_4096_VCC.bin. Since then, my experience with more recent builds has been that the battery voltage measurement bit seems to work fine but the temper/humidity/press aspects stop communicating with my Domoticz. i.e. Still have good values in my device config. Having tried to update to recent builds, such as ESP_Easy_mega-20180723_test_ESP8266_4096_VCC.bin, and a few preceeding it, seem to have the same problem.

TD-er, 2018-07-31:
Just as a note.
I have been running a BMP280 (using the same plugin) and reading VCC on the same NodeMCU since I opened this issue.
It is running rock-solid stable, at least with the commits that will be used in tonight's build.
The values are all reported fine to Domoticz, via MQTT and also Domoticz HTTP (both different IDX)

robupnorth01, 2018-08-01:
I'll check my unit when I get back from work tonight. I usually have it in sleep mode, but will check it out either way.

TD-er, 2018-08-01:
Ah, sleep mode, that changes things a bit...

robupnorth01, 2018-08-01:
Just uploaded the latest firmware ESP_Easy_mega-20180801_test_ESP8266_4096_VCC and turned off sleep-mode. Unfortunately, Domoticz doesn't get any updates for the BME280 device T, H and P. ...and the device THP values (within my ESPEasy device) don't change either, despite warming them up. VCC device does still update Domoticz and guess is still working but battery voltafe slow to change.
Hope this helps...

DittelHome, 2018-08-08:
Today (08.08),i have testet the new build. No values from BME 280, also distance Sensor. ESP run without sleep mode...

sassod, 2018-08-08:
I have the same problem with bme280 also i2c scan shows no device on all
possible addresses.

TD-er, 2018-08-08:
That's very odd, I just tested here on my test setup and I2C scan works. Also the BME280 updates fine.

Did you create new settings? (that's one thing I didn't do on my test board)

DittelHome, 2018-08-08:
I did not create new settings.
But:
DHT22 and BME280 works after submit the device definition again. After reboot, this both devices needs here trigger again (submit). In other words, after every reboot, you have to do a submit again. Maybe it depends on the interval (1800 seconds).
HC-SR04 does not deliver any reading, i deleted the device definition (task1) and  inserted again, without success.

TD-er, 2018-08-08:
Hmm that may be something indeed, that it will wait for the first interval to be passed.

DittelHome, 2018-08-08:
HC-SR04 doesnt work ..

TD-er, 2018-08-08:
The BME280 issue should be solved now.
There is still an issue with the 1st MQTT message being sent after deep sleep. That one gets lost.

Will look into that later this week (not sure about tomorrow).
It will be included in tomorrows build.

closed 2018-08-08 by TD-er